Next-Flip/Momentum-Firmware #191: Wiki

gameking2011, 2024-08-23:
### Description of the feature you're suggesting.

Add a wiki?

Willy-JL, 2024-08-23:
i have my hands full enough already. would be nice to have a dedicated wiki, but its very low priority.

gameking2011, 2024-08-23:
k though what does vgm options do???

Willy-JL, 2024-08-23:
VGM is video games module, you can buy it from flipper zero shop. if you flash RGB firmware from apps > gpio > vgm > vgm tool, then you can customize the colors of the video output of the vgm by using momentum > misc > vgm options

GameLord2011, 2025-01-24:
💡 You can just publish the docs folder as documentation.

jaylikesbunda, 2025-01-24:
> 💡 You can just publish the docs folder as documentation.

most of it is just official flipper documentation i don't really see the point

956MB, 2025-04-06:
The wiki is live (still WIP)!

closed 2025-04-06 by 956MB