Next-Flip/Momentum-Firmware #197: BIP39 crypto wallet passphrase does not save

Clats97, 2024-08-30:
### Describe the bug.

After installing the crypto wallet and setting it up, when you open the app after everything loads it says "Reccomendation: Set BIP39 passphrase".

After going into settings and setting one, and you close the app and open it again, it still says you need to set a BIP39 passphrase and when you go to settings it says the passphrase is off. 

### Reproduction

1. Turn on flipper
2. Navigate to crypto wallet app
3. Go to settings
4. Turn on BIP39 passphrase
5. Set the passphrase
6. Save the passphrase
7. Close crypto wallet app
8. Reopen crypto wallet app
9. You should see that it tells you once again to set a BIP39 passphrase
10. Go to settings, it should say BIP39 passphrase is off

xtruan, 2024-09-03:
This is working as intended. Setting the passphrase is recommended as an additional security factor.

Clats97, 2024-09-03:
Yes but why once I set a password does it say I haven't set a password?

[signatureImage]

xtruan, 2024-09-03:
The file that stores your secret keys is encrypted on the SD card. However, the encryption routine is symmetric and is the same for every Flipper device, meaning that anyone with the app can read your secret keys if they were to use their own Flipper with your SD card.

A simple workaround to improve this situation is for you as a user to use a passphrase as an extra layer of security each time you open the app to "unlock" your real wallet. Storing this passphrase between app runs would defeat the purpose, however.

This is a compromise in terms of security vs. usability to be sure. There is a fork of my original repo where the app has been modified to use the built-in Flipper keystore, which should improve security. I haven't explored integrating that into the latest app version that's deployed with Momentum though.

closed 2024-09-03 by Willy-JL