Next-Flip/Momentum-Firmware #200: Webapp Updater failed and flipper stuck

Yanking7556, 2024-09-01:
### Describe the bug.

attempted updating from Momentum-4 to 6 just now, and webapp failed with 
"890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_STORAGE_INVALID_PARAMETER", 
and followed with, 
890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_CONTINUOUS_COMMAND_INTERRUPTED
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/resources.tar.gz
890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_CONTINUOUS_COMMAND_INTERRUPTED

The Flipper said update failed, and now is in a constant rebooting loop of "unsupported radio stack".

Please let me know how I should proceed... 

### Reproduction

Not sure...

### Target

Windows 10, Version 128.0.6613.113 (Official Build) (64-bit), Flipper MMT-4

### Logs

```Text
`vendor.4e4e859b.js:1 NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user.
a @ vendor.4e4e859b.js:1
890.065c7545.js:1 Main: Flipper connected
890.065c7545.js:1 Main: RPC started
890.065c7545.js:1 Device: gui.startScreenStreamRequest: OK
890.065c7545.js:1 Main: Fetched device info
41chunk-common.929aa92b.js:1 Uncaught TypeError: Cannot read properties of null (reading 'data')
    at S (chunk-common.929aa92b.js:1:2761)
    at chunk-common.929aa92b.js:1:8178
    at vendor.4e4e859b.js:5:178449
    at Array.forEach (<anonymous>)
    at Object.emit (vendor.4e4e859b.js:5:178437)
    at l.onmessage (chunk-common.929aa92b.js:1:1796)
890.065c7545.js:1 Updater: Update started
890.065c7545.js:1 Device: gui.stopScreenStreamRequest: OK
890.065c7545.js:1 Updater: Region provisioning message: {"countryCode":"CA","bands":[{"start":304100000,"end":321950000,"powerLimit":12,"dutyCycle":50},{"start":433050000,"end":434790000,"powerLimit":12,"dutyCycle":50},{"start":915000000,"end":928000000,"powerLimit":12,"dutyCycle":50}]}
890.065c7545.js:1 Updater: Set Sub-GHz region: CA
890.065c7545.js:1 Updater: Downloaded firmware from https://up.momentum-fw.dev/firmware/mntm-006/flipper-z-f7-update-mntm-006.tgz
890.065c7545.js:1 Updater: Loading firmware files
890.065c7545.js:1 Updater: storage.mkdir: /ext/update/f7-update-mntm-006
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/firmware.dfu
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/firstboot.bin
890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_STORAGE_INVALID_PARAMETER
log @ 890.065c7545.js:1
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/radio.bin
890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_CONTINUOUS_COMMAND_INTERRUPTED
log @ 890.065c7545.js:1
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/resources.tar.gz
890.065c7545.js:1 Updater: RPC error in command 'storage.write': ERROR_CONTINUOUS_COMMAND_INTERRUPTED
log @ 890.065c7545.js:1
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/splash.bin
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/update.fuf
890.065c7545.js:1 Updater: storage.write: /ext/update/f7-update-mntm-006/updater.bin
890.065c7545.js:1 Updater: Loading update manifest
890.065c7545.js:1 Updater: system.update: OK
890.065c7545.js:1 Updater: Rebooting Flipper
890.065c7545.js:1 Main: Flipper has been disconnected
890.065c7545.js:1 Main: Flipper connected
2890.065c7545.js:1 Main: Flipper has been disconnected
890.065c7545.js:1 Main: Flipper connected
890.065c7545.js:1 Main: Flipper has been disconnected
890.065c7545.js:1 Main: Flipper connected`
```

Yanking7556, 2024-09-01:
I was able to break out of the update loop with qFlipper and a few reboot attempts. I'm not sure how to update anymore...

Willy-JL, 2024-09-01:
if that fails, please enter dfu mode (hold back and left, release back) and click repair in qflipper. after that, install momentum again

Yanking7556, 2024-09-01:
> if that fails, please enter dfu mode (hold back and left, release back) and click repair in qflipper. after that, install momentum again
Thank you for the swift reply Willy! The qFlipper app was somehow able to complete the install. I thought maybe my SD card had corrupted somehow....

Here is the full successful log as well in case it helps..

`473 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T04:40:31
473 [APP] OS info: Windows 10 Version 22H2 10 10.0.19045 Qt 6.4.2
657 [DBG] Trying serial port  at \\.\COM1
657 [DBG] Trying serial port FLIP_ROOBUT at \\.\COM3
657 [DBG] Using  serial port FLIP_ROOBUT at \\.\COM3
657 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
659 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
669 [RPC] Starting RPC session...
1916 [RPC] Failed to start RPC session: Failed to open serial port: Access is denied.
1917 [REG] Device initialization failed: Protobuf session error: Failed to open serial port: Access is denied.
15335 [REG] Detected new device: **REMOVED**
15338 [DBG] Trying serial port  at \\.\COM1
15338 [DBG] Trying serial port FLIP_ROOBUT at \\.\COM3
15338 [DBG] Using  serial port FLIP_ROOBUT at \\.\COM3
15338 [RPC] Starting RPC session...
15409 [RPC] RPC session started successfully.
15410 [RPC] (1) System Protobuf Version START
15464 [DBG] Detected protobuf version: 0.22
15464 [RPC] (1) System Protobuf Version SUCCESS
15464 [RPC] (2) Property Get START
15617 [RPC] (2) Property Get SUCCESS
15617 [RPC] (3) Storage Info @/ext START
15670 [RPC] (3) Storage Info @/ext SUCCESS
15670 [RPC] (4) Storage Stat @/ext/Manifest START
15725 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
15725 [RPC] (5) System Get DateTime START
15777 [DBG] Flipper time skew is 148038 milliseconds
15777 [RPC] (5) System Get DateTime SUCCESS
15778 [RPC] (6) System Set DateTime START
15830 [RPC] (6) System Set DateTime SUCCESS
15830 [RPC] Stopping RPC session...
15851 [RPC] Attempting to unload protobuf plugin...
15851 [RPC] Cannot unload protobuf plugin. This is probably OK.
15851 [RPC] RPC session stopped successfully.
15851 [REG] Registering the device
15851 [DEV] Version: mntm-003 commit: 186bc960 radio: 0.0.0
15851 [BKD] Current device changed to "**REMOVED**"
15951 [RPC] Starting RPC session...
16026 [RPC] RPC session started successfully.
16027 [RPC] (1) Gui Start ScreenStream START
16082 [RPC] (1) Gui Start ScreenStream SUCCESS
127468 [RPC] Serial connection was lost.
127468 [RPC] Stopping RPC session...
127469 [RPC] Attempting to unload protobuf plugin...
127469 [RPC] Cannot unload protobuf plugin. This is probably OK.
127469 [RPC] RPC session stopped successfully.
127486 [REG] Device disconnected: **REMOVED**
127486 [BKD] Last device was disconnected
159567 [REG] Detected new device: **REMOVED**
159569 [DBG] Trying serial port  at \\.\COM1
159569 [DBG] Trying serial port FLIP_ROOBUT at \\.\COM3
159569 [DBG] Using  serial port FLIP_ROOBUT at \\.\COM3
159569 [RPC] Starting RPC session...
159637 [RPC] RPC session started successfully.
159637 [RPC] (1) System Protobuf Version START
159693 [DBG] Detected protobuf version: 0.22
159693 [RPC] (1) System Protobuf Version SUCCESS
159693 [RPC] (2) Property Get START
159843 [RPC] (2) Property Get SUCCESS
159844 [RPC] (3) Storage Info @/ext START
159898 [RPC] (3) Storage Info @/ext SUCCESS
159898 [RPC] (4) Storage Stat @/ext/Manifest START
159952 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
159952 [RPC] (5) System Get DateTime START
160004 [DBG] Flipper time skew is -1189 milliseconds
[... 85 lines cut ...]
588371 [RPC] (1) System Protobuf Version SUCCESS
588371 [RPC] (2) Property Get START
588539 [RPC] (2) Property Get SUCCESS
588539 [RPC] (3) Storage Info @/ext START
588598 [RPC] (3) Storage Info @/ext SUCCESS
588598 [RPC] (4) Storage Stat @/ext/Manifest START
588652 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
588652 [RPC] (5) System Get DateTime START
588705 [DBG] Flipper time skew is -6890 milliseconds
588705 [RPC] (5) System Get DateTime SUCCESS
588705 [RPC] (6) System Set DateTime START
588759 [RPC] (6) System Set DateTime SUCCESS
588761 [RPC] Stopping RPC session...
588783 [RPC] Attempting to unload protobuf plugin...
588783 [RPC] Cannot unload protobuf plugin. This is probably OK.
588783 [RPC] RPC session stopped successfully.
588783 [REG] Device went back online
588783 [DEV] Version: mntm-006 commit: c5f6d849 radio: 1.20.0
588883 [RPC] Starting RPC session...
588950 [RPC] RPC session started successfully.
588951 [UTL] Start Update @**REMOVED** SUCCESS
588951 [DEV] Full Update @**REMOVED** SUCCESS
600230 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
600249 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
`

closed 2024-09-01 by Yanking7556