Next-Flip/Momentum-Firmware #204: Stuff In Internal storage

gameking2011, 2024-09-02:
### Describe the bug.

The latest update did away with internal storage usage, But the desktop settings and u2f key keep generating in there

### Reproduction

1. Turn on
2. Open qFlipper
3. check internal storage

### Target

Internal Storage

### Logs

```Text
qFlipper:
579 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T04:40:31
580 [APP] OS info: Windows 11 Version 23H2 11 10.0.22631 Qt 6.4.2
739 [DBG] Trying serial port 00E04C000001 at \\.\COM3
739 [DBG] Trying serial port 00E04C000001 at \\.\COM4
739 [DBG] Trying serial port 00E04C000001 at \\.\COM10
739 [DBG] Trying serial port 00E04C000001 at \\.\COM11
739 [DBG] Trying serial port FLIP_OPRAT at \\.\COM5
739 [DBG] Using  serial port FLIP_OPRAT at \\.\COM5
739 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
739 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
757 [RPC] Starting RPC session...
854 [RPC] RPC session started successfully.
854 [RPC] (1) System Protobuf Version START
918 [DBG] Detected protobuf version: 0.23
918 [RPC] (1) System Protobuf Version SUCCESS
919 [RPC] (2) Property Get START
1071 [RPC] (2) Property Get SUCCESS
1071 [RPC] (3) Storage Info @/ext START
1135 [RPC] (3) Storage Info @/ext SUCCESS
1135 [RPC] (4) Storage Stat @/ext/Manifest START
1202 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
1202 [RPC] (5) System Get DateTime START
1264 [DBG] Flipper time skew is -1989 milliseconds
1264 [RPC] (5) System Get DateTime SUCCESS
1265 [RPC] (6) System Set DateTime START
1336 [RPC] (6) System Set DateTime SUCCESS
1337 [RPC] Stopping RPC session...
1352 [RPC] Attempting to unload protobuf plugin...
1352 [RPC] Unloaded protobuf plugin.
1352 [RPC] RPC session stopped successfully.
1353 [REG] Registering the device
1353 [DEV] Version: mntm-005 commit: a2903f57 radio: 1.20.0
1353 [BKD] Current device changed to "Oprat"
1466 [RPC] Starting RPC session...
1533 [RPC] RPC session started successfully.
1535 [RPC] (1) Gui Start ScreenStream START
1614 [RPC] (1) Gui Start ScreenStream SUCCESS
2635 [RPC] (2) Storage Info @/ext START
2717 [RPC] (2) Storage Info @/ext SUCCESS
2727 [RPC] (3) Gui Stop ScreenStream START
2798 [RPC] (3) Gui Stop ScreenStream SUCCESS
3752 [RPC] (4) Storage List @/ext START
3985 [RPC] (4) Storage List @/ext SUCCESS
20717 [RPC] (5) Storage List @/ext/.tmp START
20825 [RPC] (5) Storage List @/ext/.tmp SUCCESS
22552 [RPC] (6) Storage List @/ext/.tmp/ios START
22639 [RPC] (6) Storage List @/ext/.tmp/ios SUCCESS
23405 [RPC] (7) Storage List @/ext/.tmp START
23485 [RPC] (7) Storage List @/ext/.tmp SUCCESS
24388 [RPC] (8) Storage List @/ext/.tmp/lab START
24456 [RPC] (8) Storage List @/ext/.tmp/lab SUCCESS
25188 [RPC] (9) Storage List @/ext/.tmp START
25289 [RPC] (9) Storage List @/ext/.tmp SUCCESS
26023 [RPC] (10) Storage List @/ext/.tmp/mntm START
26102 [RPC] (10) Storage List @/ext/.tmp/mntm SUCCESS
34266 [RPC] (11) Storage List @/ext/.tmp START
34366 [RPC] (11) Storage List @/ext/.tmp SUCCESS
35601 [RPC] (12) Storage List @/ext START
35815 [RPC] (12) Storage List @/ext SUCCESS
47461 [RPC] (13) Storage List @/ext/update START
47591 [RPC] (13) Storage List @/ext/update SUCCESS
54570 [RPC] (14) Storage List @/ext/update/f7-update-mntm-005 START
54688 [RPC] (14) Storage List @/ext/update/f7-update-mntm-005 SUCCESS
57721 [RPC] (15) Storage List @/ext/update START
57853 [RPC] (15) Storage List @/ext/update SUCCESS
58590 [RPC] (16) Storage List @/ext START
58795 [RPC] (16) Storage List @/ext SUCCESS
68584 [RPC] (17) Storage List @/ext START
68841 [RPC] (17) Storage List @/ext SUCCESS
```

### Anything else?

I have a custom asset pack I pieced together called N3rd
qFlipper Screenshots:
[image]
[image]

Willy-JL, 2024-09-02:
your logs literally say `1353 [DEV] Version: mntm-005 commit: a2903f57 radio: 1.20.0`

the update that "did away with internal storage" is mntm-006

gameking2011, 2024-09-02:
Oh...

gameking2011, 2024-09-02:
Wait when why do I have the .int Folder?

Willy-JL, 2024-09-02:
probably installed official firmware in between. just update to 006

closed 2024-09-08 by gameking2011