Next-Flip/Momentum-Firmware #208: Can't power off flippa without SD in some cases

htotoo, 2024-09-05:
### Describe the bug.

If there is no sd or only unmountable, then the power off function (hold back button) shows just "Update app" error.
Video: https://discord.com/channels/1211622338198765599/1211622339217989634/1281130968605786172

### Reproduction

Update to mntm 7. (in one of the dev releases just before 7 the bug was also present)
Remove SD, reboot. 
Try to power off.

I-Flip-Zeros, 2024-09-05:
https://github.com/orgs/Next-Flip/projects/1?pane=issue&itemId=56117501

Willy-JL, 2024-11-11:
fixed in [mntm-008](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-008)

closed 2024-11-11 by Willy-JL