Next-Flip/Momentum-Firmware #210: Where is the walkie talkie feature??

Ligerbot, 2024-09-11:
### Describe the bug.

Uhh I updated my flipper and there is nothing about the walkie talkie thing they added.

### Reproduction

Update your flipper

ToxxikHalo, 2024-09-11:
I think it's just for the official firmware right now. That's what the notification you got today was for... I'm sure willy can give u info on porting it over to momentum though...

Willy-JL, 2024-09-11:
There is no walkie-talkie feature, in no firmware. This was badly communicated. They mean subghz > read > config > sound ON. This has been in ALL firmwares for 2+ years. The blog post does NOT talk about new features, it is a *recap* of the last 2 years of flipper development. NOTHING from the blog post is new. 1.0.1 does contain some small features and improvements, like all updates, but NONE of them are in the blog post, and EVERYTHING in the blog post is old news.

Willy-JL, 2024-09-11:
copy paste from discord, 5th time now.

## 1.0 Firmware in Momentum??????
1. All features described in the [blog post about 1.0](<https://blog.flipper.net/released-firmware-1/>) have been around for a long time:
  - App catalog is from [August 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/2959>)
  - 3x faster NFC refactor is from [October 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/3050>)
  - 1 month battery life is from [April 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/2569>)
  - JS support is from [February 2024](<https://github.com/flipperdevices/flipperzero-firmware/pull/3286>)
  - 40% faster install is from [June 2024](<https://github.com/flipperdevices/flipperzero-firmware/pull/3716>)
  - External CC1101 support is from [June 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/2747>)
  - Faster BLE connection is from [May 2024](<https://github.com/flipperdevices/flipperzero-firmware/pull/3644>)
  - Sub-GHz sound support is from [December 2022](<https://github.com/flipperdevices/flipperzero-firmware/pull/2131>)
  - BinRAW protocol is from [February 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/2322>)
  - New IR universal design is from [August 2023](<https://github.com/flipperdevices/flipperzero-firmware/pull/3006>)
  - External IR blaster support is from [March 2024](<https://github.com/flipperdevices/flipperzero-firmware/pull/3484>)
2. All of the above changes, which is everything talked about in the blog post, are already part of Momentum stable releases, and have been for several months.
3. The 1.0.1 update however contains some routine improvements just like any other update, but it is *not* what the blog talks about.
4. All the changes that are actually new in 1.0.1 are like any other routine upgrade. This blog post is not about "what is new in 1.0" but "we have come a long way over the last 2 years".
5. Nevertheless, the actual new changes of 1.0.1 (*not* what the blog talks about, but what the actual update has new) is all already in Momentum dev builds.

Willy-JL, 2024-09-12:
leaving "open" for more visibility

Willy-JL, 2024-11-11:
closing as hopefully we are past the shitstorm xD

closed 2024-11-11 by Willy-JL