Next-Flip/Momentum-Firmware #218: mntm-007 | High power consumption at idle

0d2f1x, 2024-09-16:
### Describe the bug.

Power consumption when idle is about 9 mA. Also, Flipper does not enter "napping mode". 
P.S "Debug" is turned off

### Reproduction

1. Go to main screen
2. Hold the RIGHT button
3. Wait until the screen backlight goes out

torik988, 2024-09-16:
hey i just finish install momentum007 in stable release, like ten minutes ago and for me it works great !

brneor, 2024-09-16:
Saw the exact same behavior but everything was fine after a reboot.

0d2f1x, 2024-09-16:
> Saw the exact same behavior but everything was fine after a reboot.

Rebooting solved the problem 👍

Willy-JL, 2024-09-16:
You most likely had settings > system > debug on or sleep method legacy. Both of these options need to be set correctly to allow deep sleep (debug off sleep method default). After changing those parameters, often a reboot of a full shutdown and power on is required for the hardware to notice the change

closed 2024-09-16 by Willy-JL