Next-Flip/Momentum-Firmware #219: BadKB and BadBT Keyboard Layout Bug - MNTM-007

AztecCodes, 2024-09-16:
### Describe the bug.

When I use BadKB or BadBT and select the keyboard layout which is in my case de-CH, it doesn't work correctly.
I tried other layouts too and have the same problem.
Now this bug appeared a few versions earlier, was fixed and now it is here again.

**Version:** MNTM-007.

### Reproduction

1. Start BadKB on your Flipper Zero
2. Select your keyboard layout and run the test payload
3. You will see that the symbols are wrong

I attached the text file of the demo_windows payload with de-CH.
[Hello World!.txt](https://github.com/user-attachments/files/17009869/Hello.World.txt)

closed 2024-09-16 by AztecCodes

leow149, 2024-11-24:
Did you find a solution for this? I'm having similar issues with the de-DE layout on Win11

Willy-JL, 2024-11-24:
you have no way of knowing if youre having similar issue, the original issue here did not have any details on what actually goes wrong, just "does not work correctly" which could mean 1000 different things

AztecCodes, 2024-12-03:
@leow149 @Willy-JL 

The bug only appears in the standard windows text editor. Otherwise it works completely fine.

Because of that I closed the ticket.