Next-Flip/Momentum-Firmware #220: Battery draining faster that it should

Anonymous3-a, 2024-09-16:
### Describe the bug.

When I previously used Roguemaster's firmware, the battery drained much slower, but now it drains (with no use) five percent over about 24 hours. This is much faster than I remember, the battery would drain much slower with use. I have debug mode off. This is as of commit 9c54336, which is latest as of writing.

### Reproduction

Not sure, maybe install Roguemaster firmware and then go back to Momentum? I'll do more testing.

### Target

What's the target?

### Logs

```Text
How can I get the logs?
```

### Anything else?

I'm going to test roguemaster's firmware, it might just be in my head. I'll update this when I have the battery results.

Anonymous3-a, 2024-09-16:
Actually I'm a few commits behind, I'll test anyways

Anonymous3-a, 2024-09-16:
Ok, it turns out that I never noticed the battery drain.

Willy-JL, 2024-09-17:
Read and do what I said in #218

Willy-JL, 2024-09-17:
Also, 5% in a day is not super weird. More than expected, but not bad. If deep sleep was not working, then it would last a week in total from 100%, with 5% in a day you're on track for 20 day battery, much closer to the expected 30 day battery that deep sleep achieves.

Go into settings > power > about battery, let the screen turn off and a few seconds pass, then tell what consumption you see

Anonymous3-a, 2024-09-29:
I read #218, I have debug off + sleep method default. I don't see a settings > power > about battery, I assume you mean battery info. It says "Napping... (~3mA)".

I'm about to update my flipper (I don't update very often), so maybe the battery will do better after that.

I'm assuming this means that I should reopen the issue, so I will.

Willy-JL, 2024-10-01:
Napping means it's going into deep sleep correctly. It shows the mA value even though it's likely incorrect (on ofw it's not shown when below 5mA, because below such value the gauge can report an inaccurate value. On momentum I have made it show the detected value even though it may be inaccurate). So if it says napping, it's working as intended

Anonymous3-a, 2024-10-03:
Alright. Probably bumped a setting or something, I'm gonna back my files up and wipe+reinstall just to see if it helps. Closing for now, I'll add a comment if it helps.

closed 2024-10-03 by Anonymous3-a