Next-Flip/Momentum-Firmware #221: Mutancode not in the menu

DanielAlcon, 2024-09-16:
### Describe the bug.

Mutancode is missing from the menu for adding new SubGHz codes manually.

### Reproduction

1- Sub-GHz menu
2- Add Manually
3- Search for mutancode (if I'm not mistaken it should be something like KL: mutancode)

Willy-JL, 2024-09-17:
And on what basis were you expecting it to be there? Because I've never heard of that name

DanielAlcon, 2024-09-20:
I saw it in the code and thought it was implemented but couldn't find it anywhere
[image]

ZProLegend007, 2024-09-21:
The code snippet you've showed is from the keeloq protocol. Just use that, it's obviously in there.

Willy-JL, 2024-09-21:
Not sure if that would work, to add manually I believe you need the manufacturer's keys. KeeLoq can be detected easily, but without the manufacturer's encryption keys, which are different for each KeeLoq implementation, you can't tell what the counter is and how to make a new remote (add manually)

Willy-JL, 2024-10-30:
closing as this is not posible to implement without manufaturer keys

closed 2024-10-30 by Willy-JL