Next-Flip/Momentum-Firmware #223: Update stuck at 99 %

shivagnihotri, 2024-09-16:
Updating from official firmware 1.0.1 to mntm-007 keeps staying stuck at 99 %. What to do? 🙏

Anonymous3-a, 2024-09-16:
Where is it stuck? Can you give a picture?

shivagnihotri, 2024-09-17:
[image]

Yes. It just stays at 99 % forever.

Choder7, 2024-09-17:
I updated though WebUpdater and it worked fine, if you have the option, maybe try that.

Willy-JL, 2024-09-17:
Try restarting the flipper by holding left and back a couple seconds. Also format the sd card if it keeps not working, backup files that you need first though

shivagnihotri, 2024-09-18:
Problem solved! Thank you all for your assistance. The transmission finally worked quick and fast with [qFlipper](https://flipperzero.one/update). Momentum installing as I type. Excitement!

Thank you to all developers here. You're awesome. 🥰

closed 2024-09-18 by shivagnihotri