Next-Flip/Momentum-Firmware #226: Play Sound for Find My Flipper

Marto0o04, 2024-09-18:
### Describe the enhancement you're suggesting.

The Find My Flipper app is working perfectly but I am just wondering if it is possible to somehow use the "Play a Sound" button from the Apple's "Find My" app just like we can use it from the Flipper Mobile App

Choder7, 2024-09-18:
In case you didn't know you can do that in the Flipper Mobile App.

Marto0o04, 2024-09-18:
> In case you didn't know you can do that in the Flipper Mobile App.

I mention it in the ticket at the end it's just that I share the Flipper location with my family's phones and it will be nice to do it from "find my" in case I use another device where I do not have the flipper app

Choder7, 2024-09-18:
> > In case you didn't know you can do that in the Flipper Mobile App.
> 
> I mention it in the ticket at the end it's just that I share the Flipper location with my family's phones and it will be nice to do it from "find my" in case I use another device where I do not have the flipper app

Oh sry I don't know how I missed that...