Next-Flip/Momentum-Firmware #227: Flipper Crashing in BLESpam

Choder7, 2024-09-22:
### Describe the bug.

When clicking on Model Code in Android Device Connect in the Ble Spam App, the Flipper crashes saying it ran out of Memory.

### Reproduction

1. Open Ble Spam App on your Flipper
2. Go to 6.Android Device Connect
3. Hold OK
4. Go to Model Code
5. Press OK
6. The Flipper should Crash saying: Out of Memory

### Anything else?

Probably happens cause there are a lot of devices loaded at the same time, and there isn't enough RAM.

Possible solutions: 
- unload other apps and Homescreen that take up RAM
- don't load the entire list of devices, but only the ones visible and a few before and after the visible ones

AztecCodes, 2024-09-22:
Yes I can reproduce this bug too.

Also in BLE Spam in Vibrate 'em All:

When you try to select a Toy Mode it automatically switches to "Bruteforce".

**Reproduce:**
1. Open BLE Spam Application
2. Go to Vibrate 'em All
3. Hold OK
4. Go to Toy Mode
5. Click OK
6. Select wanted Toy Mode - Example "Classic 1"
7. Toy Mode should turn to "Bruteforce".

This happens for every Toy Mode besides "Random".

However it works fine if you change the Toy Mode by using the arrow buttons.

srmrmax, 2024-09-22:
The same problem 

Sometimes it's works when I connect and disconnect the Bluetooth from options but no always

AztecCodes, 2024-09-22:
@srmrmax 
You don't have to activate Bluetooth in the Flipper Zero Settings or in the UI of Momentum to make BLE Spam work. It also works if you have Bluetooth deactivated.

Willy-JL, 2024-09-27:
@AztecCodes found the issue, was so stupid lol https://github.com/Next-Flip/Momentum-Apps/commit/7219d039d94e5ed0674e948e5623bc79a20a5f62