Next-Flip/Momentum-Firmware #228: Minor Bug: Mifare Fuzzer App shown as mifare_fuzzer.fap in file explorer

AztecCodes, 2024-09-22:
### Describe the bug.

**Description:**
Sometimes when I open the NFC Folder with the file explorer the Mifare Fuzzer app is shown as "mifare_fuzzer.fap" with a question mark as icon instead of "Mifare Fuzzer". 

There is a way how to reproduce it (described in Reproduction), however it also appears in other occasions. I couldn't find out which other occasions trigger that too yet.

**Screenshots:**

mifare_fuzzer.fap:
[image]

Mifare Fuzzer:
[image]

### Reproduction

**A way to reproduce:**
1. Open NFC Folder
2. Open Mifare Fuzzer
3. Close Mifare Fuzzer by clicking return button

Now the title should be "mifare_fuzzer.fap" with a question mark as icon.

If you leave the NFC folder and open it again it shows "Mifare Fuzzer" normally again.

AztecCodes, 2024-09-22:
When I checked the code of Mifare Fuzzer and compared it with other applications I noticed something. The variable `fap_icon_assets` is set to `images`. Now the `fap_icon` variable has the value `images/mifare_fuzzer_10px.png`.

Could this bug be fixed by setting the `fap_icon` variable to `mifare_fuzzer_10px.png`? Since the path to the icon is already defined in `fap_icon_assets` this seems to be redundant and the full path to the image doesn't seem to be necessary. When I compared other applications it seems like it works like this, but I don't know if this could lead to this bug.

I currently can't test it on my own, I don't have the time for this at the moment, but I hope that this input might help. Since this is a very minor bug I might look into it myself when I have more time.

**The Code of application.fam:**
```
App(
    appid="mifare_fuzzer",
    name="Mifare Fuzzer",
    apptype=FlipperAppType.EXTERNAL,
    entry_point="mifare_fuzzer_app",
    requires=[
        "nfc",
        "storage",
        "gui",
    ],
    stack_size=4 * 1024,
    order=30,
    resources="resources",
    fap_icon="images/mifare_fuzzer_10px.png",
    fap_category="NFC",
    fap_icon_assets="images",
    fap_author="@spheeere98 @Sil333033",
    fap_weburl="https://github.com/spheeere98/mifare_fuzzer",
    fap_version="1.4",
    fap_description="App emulates Mifare Classic cards with various UIDs to check how reader reacts on them",
)
```

Willy-JL, 2024-09-23:
The manifest likely has nothing to do with it. Only reasons why it would not parse the fap for a name and icon are:
- the app/firmware is outdated and/or the fap can't be parsed
- the fap file is open in a file handle
First one doesn't check out since you say it works fine, then it appears broken after exiting the app. It could be the second if there is some logic error keeping the file handle to the fap file open after the app has completed execution.

I did not manage to reproduce the issue, so this needs more details. Firmware version, exact steps to reproduce starting from a fresh reboot...

AztecCodes, 2024-09-23:
Thank you for your reply!

Here we go:

**Details:**
Firmware: MNTM-007 03-09-2024
Firmware Info: mntm-007 [03-09-2024] - 0291393d [72.1] <none> - [7] momentum-fw.dev
Model: FZ.1
SD Card Format: FAT32
SD Card Space Total: 29.72 GiB
SD Card Space Free: 29.16 GiB

**Reproduction:**
1. Fresh Reboot
2. Click on the OK button (the middle one)
3. Now navigate to Apps and open it
4. Look for the NFC Folder and open it
5. Look for the Mifare Fuzzer Application and open it
6. Now close the Mifare Fuzzer Application by using the return button
7. Now you should see "mifare_fuzzer.fap" with the question mark symbol

If you go out of the NFC folder and go into it again it shows Mifare Fuzzer normaly again, but by starting the Mifare Fuzzer App and closing it with the return button the bug reappears.

I hope that you can reproduce it now. If you need any more information feel free to ask me.

ZProLegend007, 2024-09-24:
Couldn't reproduce either, you could try reinstalling firmware. My SD card is formatted as exFAT too as it is slightly newer/faster as a file system than FAT32.

Willy-JL, 2024-09-27:
same, still cannot reproduce after reflashing back to 007. please try a reinstall

AztecCodes, 2024-09-28:
I reinstalled 007 and I still have the bug. 

It seems like nobody else can reproduce it. So the ticket can be closed I think.

It must be a problem on my side; I will play around a bit and if I find something out I would post it.

Anonymous3-a, 2024-09-30:
One thing you could try is deleting the apps folder and then reinstalling, to replace the apps.

Willy-JL, 2024-10-01:
Even better, format the sd card then reinstall. Make sure to backup what you need beforehand

Willy-JL, 2024-10-01:
Doubt that the fact it's exfat means much, my sd card is fat32 and I didn't get the issue, but glad it's resolved anyway :D