Next-Flip/Momentum-Firmware #232: Potential GPLv3 Violation and Website Plagiarism by FlipperCN

kalicyh, 2024-09-23:
Dear Momentum developers,

I recently discovered that a project named "FlipperCN" (https://flippercn.com) appears to be violating the GPLv3 license of your project. They have seemingly copied parts of your code but have not made their modifications public as required by the GPLv3. In addition, their website appears to directly plagiarize your official website (https://momentum-fw.dev/), including design and content.

Here are the issues I have identified:
1. **Code plagiarism**: FlipperCN seems to have reused parts of your source code without providing proper attribution or releasing their modifications as open source, which is a clear violation of the GPLv3 license.
2. **Website plagiarism**: The design and content of FlipperCN's website closely resemble those of your website, including sections of text and images.

I have attached screenshots comparing their website to yours, along with examples of similar code sections for your reference. I believe this issue should be addressed to protect your intellectual property and maintain the integrity of the open-source community.

Thank you for your attention to this matter.

| Evidence | Momentum | FlipperCN |
|----------|----------|----------|
| Github | https://github.com/Next-Flip/Momentum-Firmware | https://github.com/ZhaiRenGaiZaoJia/FlipperZero-CN-Firmware |
| Website | https://momentum-fw.dev/ | https://flippercn.com |
| Screenshot | [image] | [image]|
| | [image] | [image] |
||[image]|[image]|
||[image]|[image]|
|||[image]|
|||[image]|
|||[image]|

WXLN, 2024-09-23:
Dogs and mice are meddling in other people's affairs.
However, in reality, the author just doesn't want to give you his code, so you want to deliberately smear and force third-party authors to open source the code. Open source agreements don't mean you can make any choices on behalf of others

kalicyh, 2024-09-24:
> Dogs and mice are meddling in other people's affairs.
> However, in reality, the author just doesn't want to give you his code, so you want to deliberately smear and force third-party authors to open source the code. Open source agreements don't mean you can make any choices on behalf of others

Moreover, I have better code, https://github.com/kalicyh/Momentum-Firmware, so why use his code?

kalicyh, 2024-09-24:
> Dogs and mice are meddling in other people's affairs.
> However, in reality, the author just doesn't want to give you his code, so you want to deliberately smear and force third-party authors to open source the code. Open source agreements don't mean you can make any choices on behalf of others

What you're saying completely misunderstands the core spirit of the GPL 3.0 open-source license. Under the GPL 3.0, any work that modifies and redistributes GPL-licensed code must follow the same open-source principles, meaning the author is obligated to provide the source code when publishing derivative works. This is not "meddling" or "smearing" but a part of the transparency and sharing spirit that protects the open-source community. If you choose to use GPL-licensed code, you must respect and comply with the GPL terms, rather than unilaterally deciding whether or not to release modified code.

WXLN, 2024-09-24:
是什么原因你自己。明白。不想把你们的恶心事放到台面上。在国内搞不定，跑到外网来搞，如果你们的脑子和心思放在代码研究上，就不会惦记别人手上的东西了。傻逼

kalicyh, 2024-09-24:
> 是什么原因你自己。明白。不想把你们的恶心事放到台面上。在国内搞不定，跑到外网来搞，如果你们的脑子和心思放在代码研究上，就不会惦记别人手上的东西了。傻逼

I don’t need anything from anyone else. My issue is with people violating the GPL 3.0 open-source license. If you had spent more time understanding the principles of open-source and respecting its terms, you wouldn’t be making baseless accusations. Trying to hide shady behavior won’t work, no matter where you do it—here or anywhere else.

Nesriaaa, 2024-09-24:
> Moreover, I have better code, https://github.com/kalicyh/Momentum-Firmware, so why use his code?

Based on the firmware downloaded from your link, I don’t think your firmware is of high quality. There are size differences in some fonts, and more importantly, the program runs into memory shortage issues. So, did you raise this issue to showcase your open-source spirit or to highlight your contributions to the Chinese firmware? I don’t believe that translating official firmware demonstrates any particular skill or technical ability. Additionally, there are reasons for not open-sourcing code, and forcing others to open-source has its own motives. Human genes are marked by selfishness and greed, which you perfectly embody. In summary, it’s like a dog catching mice—meddling in others’ business.

ZProLegend007, 2024-09-24:
> Based on the firmware downloaded from your link, I don’t think your firmware is of high quality. There are size differences in some fonts, and more importantly, the program runs into memory shortage [...]

You are a fake account that was literally made 1h ago, likely for the purpose of giving support to your other account. Chill out bro.

kalicyh, 2024-09-24:
> Based on the firmware downloaded from your link, I don’t think your firmware is of high quality. There are size differences in some fonts, and more importantly, the program runs into memory shortage [...]

First, I admit that the quality of the code is not very high, but it is based on the GPL 3.0 open-source license. I feel it is my obligation to make it public and contribute to the open-source community. Additionally, I have clearly indicated that this firmware is customized for my personal needs. I prefer to teach others how to modify and localize the firmware, rather than monopolize such knowledge. Anyone can propose better approaches, or even replace mine. However, if someone doesn't comply with the GPL 3.0 open-source license, it will only diminish the Flipper Zero ecosystem. I am happy to adhere to the GPL 3.0 license and hope more people will understand its value.

Willy-JL, 2024-09-27:
i heard from zhairen directly thanks to torron/zinongli mediating and getting us in touch. the situation seems more complicated than you make it out to be here. theres no official means of obtaining a real flipper zero in china, and many fakes being sold as "real", zhairen doesnt want to opensource his fork of momentum to prevent these fake flipper sellers from removing the warning that the flipper is counterfeit and so to protect buyers from being scammed with fake flippers. while this is not a valid excuse for violating gpl3, it is also not him trying to gatekeep the source for personal gain. ultimately, theres not much i can do here, and i dont need this drama around my project.
as for the similarities in the website design, he said it is because he admires what we built in momentum and doesnt want to take attention away from it, but rather popularize it and the only credit he wants to himself is translation.
all in all, i am not ok with him violating gpl3 of the project and will not endorse it in any way, but i will not intervene either. if someone else from the broader flipper community, since gpl3 is not only on momentum but all flipper firmware related projects, takes more of a problem with what zhairen is doing, then they can act out of their own initiative.

closed 2024-09-27 by Willy-JL