Next-Flip/Momentum-Firmware #233: Create WiFi with password, if password is entered from a device, you see it on your Flipper

Choder7, 2024-09-23:
### Description of the feature you're suggesting.

I just thought of another WiFi attack method, but I didn't seem to find it anywhere, if it already exists or isn't possible, please inform me, thanks! 👍

The Attack would probably be in Marauder, you would be able to add new ssid, select it, (already possible), and then create a wifi with password authentication,  you could name it like another WiFi or choose an ap that's already existing, or just choose all ap's, then it would create a wifi with password (or multiple).

If anybody connects to it typing in their password, you should see it on the Flipper. You should also be able to combine this with the deauth attack, so that every wifi is getting desired, while there are duplicates for each wifi hosted from the Flipper, the targets will think somethings wrong with their router, see that there's two of his WiFi's, connect to the fake one because they can't connect to the real and think it's maybe a bug of their phone, because they see every WiFi doubled, after trying to connect to the fake WiFi hosted from the Flipper, the phone requests a password, after entwring their WiFi Password, we have the targets password and the name of the router he was trying to comnect to.

Willy-JL, 2024-09-23:
I'm not super familiar with how WiFi handshakes work at a low level, but I'm pretty sure the plaintext password is not sent over the air, instead the password is hashed, and the hash is sent. So, you can't see what password the user typed. Stick to successful sniff pmkid attacks.

closed 2024-09-23 by Willy-JL

ZProLegend007, 2024-09-24:
Something similar to this is usually implemented by hosting a web server for a "login" page of an open network. That webpage would then ask you to enter a password to login (or a username and password, there can be many variations) and that password can be sent as plaintext to the flipper. However it's unlikely that the flipper had the hardware capabilities to do this and it would take a lot of work.

Helios343, 2024-09-24:
I think what your describing is called a Evil Twin rather than anything
currently available in marauder like the evil portal and such, however
using evil portal this can already be achieved with a bit of recognisance
and social engineering  on your target network and setting the portal as
there router login page to acquire there router login  or if it's the wifi
pass your alfter by same deal but maybe for example setting the portal as
same brand/manufacturer as router and making some sort of prompt for the
wifi pass to upgrade there security or something that will get them to
believe it enough to fool the target into typing there password into your
evil portal revealing it straight to the flipper in plain text you could
add a loading bar alfter they type the pass in to make it look more legit
and you close the portal they reconnect to there original network not
having any clue of what's just happend, anyway just thought I would share
to point out that this isn't a firmware based thing to be added it's got to
be done by user using the tools provided already you will see it can
already be achieved

Choder7, 2024-10-08:
How do you do the attack with the hash @ZProLegend007 mentioned? 🤔❓

Willy-JL, 2024-10-08:
@Choder7 evil portal, flash any esp32 board with marauder using the esp flasher app and then use either marauder app or evil portal app to load an html, set a name, and start the portal.