Next-Flip/Momentum-Firmware #234: repetaed bus fault & out of ram crashes

gameking2011, 2024-09-23:
### Describe the bug.

I randomly get "Bus Fault" and "Out of Ram" crashes.
How do I fix this?

### Reproduction

I can't rember.

### Target

Hardware? (idk)

Willy-JL, 2024-09-23:
Flipper hardware is very limited and we are pushing it to its limits. Simply, it's running out of ram. What you can try is changing what asset pack your using, since asset packs containing icons and fonts will consume some ram in background. Also some third party apps are coded a bit poorly and will leak memory, so after a while of using a flipper, a out of memory crash is normal. If you can reproduce a crash consistently, and disabled asset packs, then you might have a bug on your hands by we can try to fix.

gameking2011, 2024-09-24:
k
it usually happens when using the nfc app

Helios343, 2024-09-24:
> Yes the issue with the NFC App is known. It was already reported.
>
> Check the issue here: DarkFlippers/unleashed-firmware#778
>
> Using the Flipper Zero and then starting the NFC App can lead to Bus Fault
> or Out of RAM Errors.

Ohhh I see if your using one of those modules just be aware that that they
can short out on the antenna connection there onto the board

FalsePhilosopher, 2024-09-24:
Hold your back arrow for the quick power off function, does it display the crying dolphin or just turn off power? Yesterday I upgraded from 75e to 78e and the issue persisted. I noticed the different power off menu and went about fixing it. Rebooted into DFU mode, repaired to 1.0.1 from qflipper, once it booted after install I formatted the sd card. I then installed 78e from qflipper and I got my crying dolphin back an it doesn't crash launching files from explorer. I have not done extensive testing, just launched a file from browser and micro python which crashed with a bus error the first time but launched the second time. I came to the UL issue to report my findings and figured I would mirror them here as well.

gameking2011, 2024-09-25:
r u using dev? im using release.

gameking2011, 2024-09-25:
wait that is ofw

gameking2011, 2024-09-25:
that's an ofw release @FalsePhilosopher

gameking2011, 2024-09-25:
1.0.1

gameking2011, 2024-09-25:
also, @Helios343 im using internal modules

Helios343, 2024-09-26:
Sorry there was a error on my end that message wasn't ment to be in this
thread

closed 2024-09-27 by Willy-JL

gameking2011, 2024-09-28:
💡 you could unload anything that isn't currently being used

Willy-JL, 2024-10-01:
Which is what happens?

gameking2011, 2024-10-01:
*Forehead slap*