Next-Flip/Momentum-Firmware #235: Installer exe for modified qFlipper app?

gameking2011, 2024-09-23:
### Description of the feature you're suggesting.

I checked and in the build steps it says it will generate an installer exe at the end of the process. can you just put the installer in the website?

gameking2011, 2024-09-23:
honestly i'd count it as a 4'th B-day present (Lego set, Disney Now shutting down, and thing from friends that's coming in a few days)

Willy-JL, 2024-09-23:
No, we will not distribute nor endorse using modified qFlipper builds. The fork is there only as a proof of concept and for personal ease of use for myself. From what I heard, it is the Flipper Devices team's wish that there be no competing flipper companion apps, and I respect that. I'm not even sure if they're ok with just the code being up, and I'm willing to take it down if they requested so.

closed 2024-09-23 by Willy-JL