Next-Flip/Momentum-Firmware #236: Sub GHz App freezes when I use Frequency Analyzer and select the found signal when Modulation FM238 is set

AztecCodes, 2024-09-24:
### Describe the bug.

**Device Information:**
Firmware: MNTM-007 03-09-2024
Model: FZ.1
SD Card Format: FAT32

**Sub GHz Settings:**
Module: Internal (Bug doesn't occur with external antenna)
Modulation: FM238
Bin RAW: ON

**The Bug:**
When I open the Sub GHz Application and start the Frequency Analyzer I send a signal and the Frequency Analyzer shows the Frequency. I click on OK Button then it reads in this frequency. At this point the Flipper freezes. It doesn't even crash, it just freezes. I have to force reboot it by using button combination. This only happens with Modulation FM238 and only with internal antenna in my case.

### Reproduction

1. Reboot Flipper Zero
2. Select Sub GHz Application in Menu (Or start it from File System)
3. Select "Read" and go to config and set Modulation to FM238. Also turn Bin RAW to ON.
4. Now use the return button to go back to the menu of the Sub GHz Application
5. Select "Frequency Analyzer"
6. Send a signal or wait till your receive a random signal so the Frequency Analyzer shows a Frequency
7. Click the OK Button so it starts the read function with the found frequency
8. At this point the Flipper should freeze. You can't do anything and you have to force reboot with button combination.

### Anything else?

I tested every Modulation and it only happens with FM238.
When I use this external antenna: [Flipper Zero External CC1101 Module](https://www.tindie.com/products/flipmodules/flipper-zero-external-cc1101-module/)
the bug doesn't appear. 

If more information is necessary feel free to ask me!

Helios343, 2024-09-24:
I noticed this too acually, what external c1101 chip are you using ? I've
got a ebyte here

Helios343, 2024-09-24:
Ohhh I see if your using one of those modules just be aware that that they
can short out on the antenna connection there onto the board

AztecCodes, 2024-09-24:
I linked the exact external antenna I used in my issue report. However the relevant point is that the bug occurs with the internal antenna.

Helios343, 2024-09-24:
Weird I'm getting the same bug with external ebyte module

AztecCodes, 2024-09-24:
This is not weird. If this bug occurs with your antenna it is good to know. Different Antennas can lead to different things I guess.

AztecCodes, 2024-10-06:
**Little update:**
After I formatted my SD Card from FAT32 to exFAT and made a complete fresh install, the problem still persists.

Choder7, 2024-10-06:
Something similar happened to me.

AztecCodes, 2024-10-08:
Relevant: [Same issue in Unleashed Repository](https://github.com/DarkFlippers/unleashed-firmware/issues/817)

AztecCodes, 2024-10-09:
Relevant: Bug still occurs on MNTM-DEV 3AA97E69

bip3r, 2024-10-30:
I confirm the bug.
It occurs regardless if I use external antenna, or internal one.