Next-Flip/Momentum-Firmware #238: NFC I2C+ 2k broken

Helios343, 2024-09-28:
### Describe the bug.

Since updating from c0d8659 to fb4d0032 my previously saved nfc i2c+ 2k reads no longer let me open info on them crashing 'out of memory' everytime, I've tried also to re-read the tags and it reads as ntag203 and data is all wrong reinstalled firmware still the same, reverted back to c0d8659 and reads fine and works normally

### Reproduction

Newest dev-release
https://github.com/user-attachments/assets/4635831d-830e-4a6e-b1ba-e047b4adf14e

### Target

NFC I2C+ 2k

Willy-JL, 2024-10-01:
The recent dev builds include recent PR #237 which parses I2C+ tags as NDEF too. Either the card contains too much data to display, or there is a bug in the NDEF parser. Kindly send a copy of the nfc dump for the card, as read on firmware that doesn't crash.

Helios343, 2024-10-11:
Working read of tag on older build

https://flpr.app/sf#path=nfc/NTAG_I2C%2B_2K_2024-09-12_12%2C09%2C33.nfc&key=fLsxe5LMv-_6l5-GHrPGzg&id=luKC2v

What it now reads as in newer firmware

https://flpr.app/sf#path=nfc/NTAG203_2024-09-18_00%2C31%2C01.nfc&key=KwBIP8-zB0tVOXjehbY51Q&id=zJOlEu

Willy-JL, 2024-10-24:
@Helios343 can you try on latest devbuild? #265 made many improvements to ndef parsing. thoug hadmittedly im not sure if that tag even has ndef, but it does sound related

Willy-JL, 2024-10-30:
for what its worth, i opened the previously read file with the new ndef parser from devbuild and it shows a bunch of ndef data, but it curiously has a wrong string length value. it has a very long string, and its last character is 0xFE which is the ndef TLV terminator. so the string length should have been 1 less, since the TLV is supposed to be outside of it. or maybe they duplicated the 0xFE TLV terminator for some reason. seems off because the rest of this string seems like plain text, but since the last character is 0xFE which is not a printable character, it prints everything as hex instead of as text