Next-Flip/Momentum-Firmware #241: Bad usb

FlipperNoob75, 2024-10-03:
Can't find the option for running bad usb thru Bluetooth

Anonymous3-a, 2024-10-04:
Go to BadUSB, open your file, press <- for config, set connection to BT

FlipperNoob75, 2024-10-04:
All I have is bad KB. There isn’t even an app for bad usb on my device and
I have the latest version

Willy-JL, 2024-10-04:
So you acknowledged that there is "Bad KB" but your reasoning is preventing you from ever opening like you are tied to a chair, just because it's not called "Bad USB" as you expected?

Survival instincts are diminishing it seems.

FlipperNoob75, 2024-10-04:
I have opened it. It takes me to an arrow pointing up and below that it
says demos. I click on the Up arrow and it shows folders which one is
called bad usb. My problem is no matter what I click bad kb or the folder
bad usb there is no config for me to select

FlipperNoob75, 2024-10-04:
I figured it out each demo has to be set to bt. I’m new to this I’ve had a
flipper less than a week and haven’t owned a pc in 20 yrs. Is there any
more bad kb files I can get from you guys?

FlipperNoob75, 2024-10-04:
I’ll figure that question out to. Appreciate the help not sure what the
attitude was about but it’s funny you mentioned survival skills diminishing
because if the world goes to shit. A computer won’t feed,water or shelter
ya. I was raised in the mountains so I have all those skills plus I can
pretty much build a house, wire & plumb it. I don’t know nothing about
computers because I was never into computers but now I figured I’d give it
a shot and with it being a week I’d say it’s a little early to judge me. We
all have things that were good at you’re good at coding board making
whatever it is that you do I like your guises firmware, but that’s your
cuppa tea not mine so I’m not any less intelligent than you are. I just
haven’t had a need for computers till now. So hopefully our next
interaction will be a little more civil. You have a good night.

closed 2024-10-07 by Willy-JL