Next-Flip/Momentum-Firmware #247: add setting to adjust charging current limit

SinKy-Yan, 2024-10-06:
### Description of the feature you're suggesting.

I'm using a DIY Flipper Zero, and its PMU inductor packaging is incorrect. It overheats at the default charging current. I'd like to add a setting to adjust the charging current in the power settings. I noticed the driver already records the corresponding register locations. I tried using ChatGPT to write code for adding this setting, but it didn't succeed.

Aholicknight, 2024-10-23:
[image]

This already exists.

Willy-JL, 2024-10-23:
no, this issue is asking for current limit, not percentage cap. it means how fast it charges.

SinKy-Yan, 2024-10-30:
Yes, because the inductor package I am using is a bit small, the inductor temperature rises to 60 degrees when charging at 600mA. I believe that after it is enclosed in the casing, the temperature will be even higher, which seems to pose a safety hazard.

RichardoC, 2025-01-23:
It looks like the official board uses BQ25896 for charging with datasheet https://www.ti.com/lit/ds/symlink/bq25896.pdf

Looks like you need to adjust the bits in REG00 in https://github.com/Next-Flip/Momentum-Firmware/blob/dev/lib/drivers/bq25896.c in bits 0-5?