Next-Flip/Momentum-Firmware #254: Add function in explorer to select multiple files

AztecCodes, 2024-10-08:
### Describe the enhancement you're suggesting.

Add the function to the explorer that you can select multiple files and then make use of action like cut, delete and copy.
This would be great!

Anonymous3-a, 2024-10-09:
I feel like if you're doing anything more complex than one or two files, you should just use a computer to move the files, although that would be a nice feature.

Choder7, 2024-10-09:
Cool Suggestion! 👍

You would probably hold down OK in the file explorer and then click on SELECT, then you can just click on other files _without having to hold down OK and choose select again for every file._ When you selected every file you want to move, copy, ..., you just hold down and click on move, copy, ...

fiveangle, 2024-10-25:
This actually would be pretty cool !

956MB, 2024-12-19:
> You would probably hold down OK in the file explorer and then click on SELECT, then you can just click on other files _without having to hold down OK and choose select again for every file._ When you selected every file you want to move, copy, ..., you just hold down and click on move, copy, ...

Are we talking about file only selection, or files and folders? Because I think the process for selecting makes sense if you only want to do files or folders, not both at the same time. Is that the main goal with multiple selection here? Like one would only really be dealing with a handful of files in a folder at a time, rather than a large selection of files/folders in different locations? I haven't had the need/desire to move multiple things around yet, but I def think it *should* be a thing.

With the hold OK being for the Actions context menu, and the click OK being for selecting items in "select mode", how do you click in to select the files in some folder, and select an entire folder? Seems like its a potentially "keybind limited" action if you're trying to do more than files, right? Am I making this too complicated or just misunderstanding?

Willy-JL, 2024-12-19:
i think having both file and folder selection would be ideal, as it reflects what you can do in desktop file managers for example. but you bring up a good point, that it has some extra concerns for inputs.

i had a couple ideas. all of them would start with holding ok, then entering select mode. then:
1. ~~press/hold ok on file to select, press ok on folder to enter it, hold ok on folder to select it~~ i realize after writing that this wouldnt work, as you wouldnt have a way to bring up the more actions menu as hold ok is already used
2. press left/right (not sure which, or if both) on file/folder to select, press ok on folder to enter it, hold ok on file/folder for actions menu, press ok on file does nothing (?)
3. press ok on file/folder to select, hold ok on file/folder for actions menu, press right on folder to enter it

2 seems most reasonable as it works with the current paradigm of ok to enter folders. maybe it could also allow pressing ok on files to select them
3 makes it less obvious how to navigate, but easier and more consistent to select

also, allowing folder selection brings another concern. before the copy / move action by pasting selected files/folders, would need to check if the destination path is a child of one of the selected items, as that could cause issues (this was a problem early on with the cut copy paste options, and i patched it in storage service to throw an error, but would still probably be wise to check for this before starting to copy or move items to fail early before any of the selected items are processed)