Next-Flip/Momentum-Firmware #268: Bluetooth Scanner

MathematicianGoat, 2024-10-15:
### Description of the feature you're suggesting.

**1.** I know that the BT stack is not enough to create a bluetooth scanner for the original firmware. Is it possible to do it in Momentum? 
**2.** Also, it would be great if you could add more example apps (ex: BT packet sniffer) about Bluetooth. Because it is very hard to find faps that set an example to develop our custom code.

### Anything else?

Thanks for the great firmware!

closed 2024-10-15 by Willy-JL

Choder7, 2024-10-15:
Just wondering if there's any way that the applications could be flashed/copied onto the SD card while updating the Flipper from the web app or something, thus giving you much more space to work with.

Willy-JL, 2024-10-15:
That's exactly what happens. The problem is the firmware space on flash. Only firmware is on flash. And it doesn't fit.

ZProLegend007, 2024-10-15:
Not super experienced with this sort of stuff, but is it possible - for advanced users perhaps - to invest in a project to have a core necessary firmware stored on the flipper flash itself, and then larger firmware files stored on the SD card that could be called on? Or is there a serious problem with that. I understand this would make the SD card ever more important but the idea is that there is still core firmware on the flash so it doesn't explode without the SD just in case.

Willy-JL, 2024-10-15:
Everything that can be put on sd card is already on sd card. If we could do anything about this it would have been done long ago.

"Move things to sd card" and then what? You can't execute code from sd card. Once it's on sd card, it must be loaded into ram to be executed. Ram is 4 times smaller than firmware flash. And as it is already, we don't have enough ram. We don't have enough flash. There is NOTHING we can do.

Willy-JL, 2024-10-15:
What you could do is go back in time and tell zhovner and the team to not use 1980 era hardware specs for the flipper.

Choder7, 2024-10-15:
Im gonna do that real quick

MathematicianGoat, 2024-10-15:
Ah, what a bummer. I thought i was the problem. Thanks to everyone for the replies

ZProLegend007, 2024-10-16:
Ok, perhaps, and bear with me... hardware mods? Soldering on a new chip with more ram? More flash?

jjones3135816, 2024-10-16:
Google the chips!Nobody makes these with more flash and ram and it”s not like you can just bolt on a 4-barrel carb and you are god to go.JJ

Willy-JL, 2024-10-16:
indeed. if you wanted to add more ram or flash, youre looking at the same kind of effort of making a fully new generation of flipper. leave that to flipperdevices