Next-Flip/Momentum-Firmware #269: Ring Airtags

Choder7, 2024-10-15:
### Description of the feature you're suggesting.

https://github.com/DarkFlippers/unleashed-firmware/issues/625

I can do it in my Android Settings, so should be possible, right?

### Anything else?

no

Choder7, 2024-10-15:
Wait, that's not the same?

Willy-JL, 2024-10-15:
What is the point of this? Airtags are not paired to the flipper

ZProLegend007, 2024-10-15:
Might be interesting to see how the app does it though, considering they aren't paired to the android devices either... if it's a Bluetooth signal it may be possible

Choder7, 2024-10-15:
They Send a Signal specifically for the point to be detected and ringed.

Willy-JL, 2024-10-15:
"Bluetooth signal" "send a signal" it's not that easy. If it requires the master device starting a connection to the airtag, you can kiss goodbye to this on flipper. If it is an advertisement packet, it could work. But what do you put in the packet? Obviously there is no packet that makes ALL airtags ring, that would be stupid and Apple is many things but not stupid. Most likely, there's a packet that tells the airtag to ring, and it contains an identifier for what airtag should ring, so only that one rings. How do you find this identifier? A phone can scan nearby devices. Flipper cannot. And you sure as hell are not typing 64+ hexadecimal characters to identify the airtag.

Willy-JL, 2024-10-15:
"But bRuTeFoRcE!!" Bruteforcing this will take 8(possible values for a byte) ^ 32(rough estimate, public keys used by airtags are even longer than 32 bytes since they use both the 31 byte adv payload and the 6 byte MAC address, but let's keep it on a lower estimate) which gives 7,922816251426e28 possible identifiers, assuming 20ms advertising rate this means 5,024617105166e19 YEARS for bruteforcing airtags.

closed 2024-10-15 by Willy-JL

Choder7, 2024-10-15:
Ok, sorry, I didn't know the Flipper couldn't scan...

Willy-JL, 2024-10-15:
Yeah, another sad limitation of having the light ble stack

Choder7, 2024-10-16:
But the the Flipper should be able to send specific Bluetooth packets, if so, you could make an app like BLE spam where you enter the entire hex code, and not just the model code which is then put in the hex code that is beeing send.

If you can edit or type the hex code yourself, you should also be able to set the TX Power (signal strength) that the Flipper tells the other ble devices like your phone.
The TX Power I think is used to determine the Range of the device in pairing mode, so that only close devices will cause a popup.
It is marked with 0a, at least with Googles Fast Pair (Android).

Just a thought of me, because while sniffing the ble package that the Flipper sends, the byte after 0a was randomized for every signal.

_jb0x168.github.io/ble-spam-analysis says:_ "There are several AD Types. Most of them, like Flags (0x01) and TX Power (0x0a) are standard types that are used by both Google and Apple protocols."

ZProLegend007, 2024-10-16:
I think I know what you mean. There is a custom Bluetooth advertisement application on backlog for mntm that perhaps could be used for this purpose but something tells me this isn't advertisement packets we are talking about.

It would require sniffing the Bluetooth signal from a different device and then manually entering the hex code on the flipper.

Choder7, 2024-10-16:
I meant a Hex code BLE Spam/Sender in general, not just for this purpose. The example from me above is meant like the BLE spam app but you can edit the entire hex, not just the model code that the app then puts into the hex with other hex code. Because then you can edit the TX Power (signal strength (not sending signal stronger, but saying with how much power the signal is sent, so that the other device knows the range of the Device that send the signal.)) or stuff other in the Hex.