Next-Flip/Momentum-Firmware #271: Unable to use my Chamberlain garage door with recent FW versions

myTerminal, 2024-10-16:
### Describe the bug.

On the final version of Xtreme FW, I could clone my garage remote and use the .sub file to operate the garage door. Since my move to Momentum FW, none of the versions have allowed me to open/close the door (using the previously generated `.sub` file), nor can I intercept signals from other physical remotes I have to clone them. I do not see the signals on the Frequency Analyzer, even though other signals pop up in the log.

### Reproduction

1. Hit the "Center" key
2. Navigate to Sub-GHz
3. Hit the "Down" key twice to select "Saved"
4. Navigate to "garage-Door.sub" file and hit the "Center" key
5. Select "Emulate"
6. Select "Send"

### Anything else?

I earlier believed it worked on the official FW, but it turns out [that was a confusion](https://github.com/flipperdevices/flipperzero-firmware/issues/3940) given how early it was for me to switch to unofficial FWs, specifically Xtreme. I was also told that it is a "dynamic" code, so maybe we have stopped supporting it since the last version of Xtreme FW?

Aholicknight, 2024-10-24:
https://github.com/user-attachments/assets/51b13abc-c795-4a09-a367-645a74853589

Works for me fine. Do you have "Momentum > Protocols > SubGHz Bypass Region" turned **ON**?

myTerminal, 2024-10-25:
> Works for me fine. Do you have "Momentum > Protocols > SubGHz Bypass Region" turned ON?

Yes, I have this turned ON. However, neither does the door opener respond to these signals, nor is the Flipper able to intercept those from the physical remote or the car's HomeLink system, though the only thing (among garage door opener, the physical remotes, the car, and the Flipper) that has changed is the FW version.

Willy-JL, 2024-10-25:
can you try enabling some of the ignore options in read config?

myTerminal, 2024-10-26:
> can you try enabling some of the ignore options in read config?

I apologize, I didn't understand that. Could you please elaborate?

Willy-JL, 2024-10-26:
Subghz > read > config > there's a bunch of options to ignore protocols you're not interested in. Ignoring these can improve detection of what you're interested in.

myTerminal, 2024-10-26:
> Subghz > read > config > there's a bunch of options to ignore protocols you're not interested in. Ignoring these can improve detection of what you're interested in.

I tried removing all of those but unfortunately, it still did not work.