Next-Flip/Momentum-Firmware #280: get the cli improvements back

GameLord2011, 2024-10-27:
### Description of the feature you're suggesting.

bring back the cli improvements part 1 commit back!

Willy-JL, 2024-10-27:
it never went away...

GameLord2011, 2024-10-27:
huh, a commit was titeld "repeal cli impovements part one"

GameLord2011, 2024-10-27:
wait, if it never went away, then i need to edit something

Willy-JL, 2024-10-27:
It was reverted in official firmware and unleashed. By how git works, when I merge the commit shows up for us too, but I did not accept the changes of this commit. You can tell that this commit did not originate in momentum because it is not made by me and it is not linked to a momentum pull Request. Because of this, you need to instead look at the merge commit that added this commit. It's something like "merge remote tracking branch ofw/dev", and if you look at the changes of such a merge commit that added the cli revert, nothing of the cli revert was kept while merging.

GameLord2011, 2024-10-27:
oh, i found neofetch in here though.

Willy-JL, 2024-10-27:
Exactly, because it was added with cli improvements part 1 and we did not revert it, while other firmwares did revert it.

closed 2024-10-27 by GameLord2011