Next-Flip/Momentum-Firmware #283: Esp32 wifi Marauder crash

MRYIYAN, 2024-10-29:
### Describe the bug.

When I scan wifi networks on the flipper it scanned the whole networks but when I want to show them on the list it crash 

### Reproduction

When I open the list option

Willy-JL, 2024-10-30:
can you try disabling your asset pack temporarily?

MRYIYAN, 2024-11-01:
Hello! I have already try that and the still the same issue

AztecCodes, 2024-11-06:
@MRYIYAN Can you add a step-by-step guide on how to reproduce the bug? Please also add information about your device (version etc.) and if you use the original WiFi Dev Board.

Also please explain with which tool did you flash Marauder?

jaylikesbunda, 2024-11-17:
> Hello! I have already try that and the still the same issue

could you try this version of the marauder app?
[esp32_wifi_marauder.zip](https://github.com/user-attachments/files/17789633/esp32_wifi_marauder.zip)