Next-Flip/Momentum-Firmware #292: Feature Request: Extra settings for external speaker

ZProLegend007, 2024-11-06:
### Description of the feature you're suggesting.

Hello again,

I'm wondering if theres any firmware tweaking that can be done to add options in the firmware for people who have modded their flippers with an external speaker. I've recently wired up a very nice little speaker from a zenbook and nuked the buzzer that was in the flipper. It sounds leagues better but I'm wondering if it's possible to increase output power via firmware (or if its a hardware limitation) or find a way to fix distortion issues in apps like video and wav player. Wav player sounds like it has a much lower audio quality than other apps - possible because all the files are amplitude boosted.

Also I've noticed that the flipper seems to use square waves for audio output? (Sounds like it at least), if this is hardware then oh well but if possible maybe this could be a toggle in firmware? Don't know if any of this is at all possible but it's always worth checking. Basically any firmware toggles that could be added that would benefit custom external speakers.

### Anything else?

Thank you!

x10102, 2025-01-12:
Looking at the schematics, it seems that the speaker is directly connected to one pin of the MCU without any amplifiers. So I believe that the output power is a hardware limitation, unfortunately. 

The flipper does use square waves for audio output, there aren't really any other options as the STM32 doesn't have an internal DAC. I guess you could connect an external amplifier circuit with a DAC chip and send the audio to it using something like I2S, but it honestly sounds like overkill to me, I'm not even sure if there's enough memory and processing power available on the Flipper to decode higher quality audio files.

(not one of the firmware devs, just sharing my thoughts)

ZProLegend007, 2025-01-15:
That makes sense. Thanks for the extra info. Probably is overkill, I'm just seeing if it's possible to use as a simple mp3 (or wav) player with that high quality speaker I've wired up to it.