Next-Flip/Momentum-Firmware #298: Flipper crashing when I’m scan Mifare plus card

dis667, 2024-11-12:
### Describe the bug.

When I’m going NFC->Read and scan MIFARE Plus card flipper crashing with an error “furi_check failed”

### Reproduction

update to momentum 007(was all fine on this firmware)

### Logs

```Text
[23:13:29][info]: disconnected
[23:13:29][debug]: reconnecting
[23:13:29][info]: connecting
[23:13:29][info]: session started
[23:13:29][info]: canceling tasks...
[23:13:29][info]: canceling tasks done
[23:13:33][info]: connected
[23:13:33][info]: synchronize
[23:13:34][debug]: > request(system(property(devinfo.hardware.target)))
[23:13:34][debug]: < response(system(property(hardware.target: 7)))
[23:13:34][debug]: < response(ok)
[23:13:34][debug]: > request(storage(stat(/ext/Manifest)))
[23:13:34][debug]: < response(storage(stat(162343)))
[23:13:34][debug]: > request(storage(read(/int/.region_data)))
[23:13:34][debug]: < response(storage(file(40 bytes)))
[23:13:34][debug]: > request(storage(info(/int)))
[23:13:34][debug]: < response(storage(info(1.2 GiB / 28.8 GiB)))
[23:13:34][debug]: > request(system(setDate(2024-11-12 20:13:33 +0000)))
[23:13:34][debug]: < response(ok)
[23:13:34][info]: syncing date: 585 ms
[23:13:34][debug]: > request(system(property(devinfo.firmware.api)))
[23:13:34][debug]: < response(system(property(firmware.api.major: 78)))
[23:13:34][debug]: < response(system(property(firmware.api.minor: 1)))
[23:13:34][debug]: < response(ok)
[23:13:34][debug]: > request(storage(info(/ext)))
[23:13:34][debug]: < response(storage(info(1.2 GiB / 28.8 GiB)))
[23:13:34][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[23:13:34][debug]: < error(StorageError: File/Directory does not exist)
[23:13:34][debug]: > request(system(property(devinfo.firmware.origin.fork)))
[23:13:34][debug]: < response(system(property(firmware.origin.fork: Momentum)))
[23:13:34][debug]: < response(ok)
[23:13:34][debug]: > request(storage(read(/ext/favorites.txt)))
[23:13:35][debug]: < response(storage(file(0 bytes)))
[23:13:35][debug]: > request(system(property(devinfo.firmware.origin.git)))
[23:13:35][debug]: < response(system(property(firmware.origin.git: https://github.com/Next-Flip/Momentum-Firmware)))
[23:13:35][debug]: < response(ok)
[23:13:35][debug]: > request(storage(write(/ext/favorites.txt, 0 bytes)))
[23:13:35][debug]: < response(ok)
[23:13:35][info]: syncing archive: 362 ms
[23:13:51][info]: synchronize
[23:13:51][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[23:13:51][debug]: < error(StorageError: File/Directory does not exist)
[23:13:51][debug]: > request(storage(read(/ext/favorites.txt)))
[23:13:51][debug]: < response(storage(file(0 bytes)))
[23:13:51][debug]: > request(storage(write(/ext/favorites.txt, 0 bytes)))
[23:13:52][debug]: < response(ok)
[23:13:52][info]: syncing archive: 144 ms
```

Septimus4, 2024-11-13:
Same crash with a Ultralight Ev1 tag password protected

Smilee95, 2024-11-13:
Same crash for Write to Initial Card

Willy-JL, 2024-11-13:
Can y'all try on official firmware 1.1?

Septimus4, 2024-11-14:
> Can y'all try on official firmware 1.1?

On 1.1.2 it works

Willy-JL, 2024-11-14:
what about on latest unleashed?

Septimus4, 2024-11-14:
I can test later.
FYI if I provide the password reading the tag  works on momentum

Septimus4, 2024-11-14:
It works on the latest unleashed

TheRealUnexpected42, 2025-03-07:
Reproduced bug on momentum 008 (11-11-2024). Tested using ofw and issue not present.

Issue occurred attempting to crack mifare plus X 2K SL1 keys by reading as mifare classic. Same crash error.
Reading card as mifare plus only reads standard ISO14443-4 data: UID, ATQA and SAK (no keys).

closed 2025-03-07 by Willy-JL