Next-Flip/Momentum-Firmware #306: subghz 387-464 MHz band suddenly bad reception

pikachu2010, 2024-11-21:
Hi guys,

I have a flipper zero for three months now.
I play a lot with the subghz tool, specially in the 387-464 MHz range.

I notice that suddenly the reception in this band is very poor as before. 
For example if i want to scan the frequency of my car key, now i have to place it almost against the flipper zero for reception.
Before I could detect the frequency meters apart. And also other remote keys in this range have the same problem.

Any idea what is happening here?

Willy-JL, 2024-11-21:
are you using an external subghz module?

pikachu2010, 2024-11-22:
no, just the flipper. Always used the flipper to capture signals in this range and now its almos sensitive in this range.

X3r0C00L, 2024-12-03:
I am having same issue. With both internal and using rabbit labs flux capacitor.  In frequency analyzer I can turn the RSSI sensitivity all the way to the left and nothing. Whereas at work I usually get inundated with multiple transmissions. Will barely pick up my keyfob from 8 inches away. Double checked all settings.

Willy-JL, 2024-12-03:
Read the changelog for 008, it has changes to frequency analyzer to prevent useless noise. What you before considered "inundated with multiple transmissions" was just useless noise that made no sense. The point of frequency analyzer is to find what frequency is used by a signal that you already know about: like press a button on a remote and see what frequency it is. Due to the nature of how the frequency switching works and band isolation is done at a hardware level, yeah the range is very small. This is not a bug, rather it was bugged before, and now it is fixed. You are just confusing the previous bugged behavior as useful when actually it was not.

closed 2024-12-03 by Willy-JL

X3r0C00L, 2024-12-03:
Okay thank you for the clarification. That will be better then, becaue when I need to listen for things from a distance I can use my HackRF portapack to scan and find local transmissions. 
Appreciate it. :)

X3r0C00L, 2024-12-03:
One last question. Would you mind and be able to tell me Which C files in furi are responsible for this change?