Next-Flip/Momentum-Firmware #307: BadKB doesn't write ñ or Ñ in spanish layout

iecxiv, 2024-11-23:
### Describe the bug.

Over bluetooth on android connection, doesn't write ñ or Ñ from string

### Reproduction

STRING ñ
STRING Ñ

Writes nothing on es_ES layout

### Target

Android

Willy-JL, 2024-11-23:
yeah, badusb scripts can only handle ascii characters. the purpose of keyboard layouts is to make it so the ascii characters map to the correct ones for the target computer, but it does not magically add support for non ascii characters.
by ascii, i mean specifically the original lower ascii range, up to character code 127, which does not include any accents or variations of letters, so it does not include ñ. if you absolutely require to type this character, you might be able to use `ALTSTRING ñ`, or `ALTCHAR 164`. please see here: https://github.com/Next-Flip/Momentum-Firmware/blob/dev/documentation/file_formats/BadUsbScriptFormat.md

closed 2024-11-23 by Willy-JL

iecxiv, 2024-11-23:
On android does not work ALTSTRING or ALTCHAR