Next-Flip/Momentum-Firmware #308: Secondary fonts are glitching.

GameLord2011, 2024-11-24:
### Describe the bug.

I am using a customized version of mntm-8 (all changes are in the cli so I'm pretty sure it isn't causing this) and the Watch dogs secondary font is not working, and it is switching to the default secondary font.

### Reproduction

1. Turn on flipper
2. change asset pack to _Watch Dogs_
3. open apps menu
4. open nfc app

### Target

Firmware

Willy-JL, 2024-11-24:
asset packs are intentionally disabled in very few select apps. nfc, mfkey and upython apps are very heavy apps and require as much ram as they can get, so disabling asset packs helps with memory issues for those apps

closed 2024-11-24 by Willy-JL

GameLord2011, 2024-11-24:
k

GameLord2011, 2024-11-24:
then what about the "nfc" and "rfid" icons in watch dogs?

Willy-JL, 2024-11-24:
:man_shrugging: its there and the nfc app doesnt use it. rfid app still uses asset packs so not sure why mention that. and the `NFC_dolphin_emulation_51x64` is actually used by the rfid app and the esubghz chat app too, so you still see it.

GameLord2011, 2024-11-24:
sry read mfkey as rfid