Next-Flip/Momentum-Firmware #315: Question

GameLord2011, 2024-12-02:
How would I change the color of the info text in the neofetch command. I have a personal fork of momentum fw, and I want the ascii art color to be set to ANSI_FG_BLUE but the color for the info text to be set to ANSI_FLIPPER_BRAND_ORANGE. the code is in GameLord2011/Momentum-Firmware._** What I tried is defining a new variable using:
```C++ 
#define 
```
 called "NEOFETCH_INFO_COLOR" and then adding it after it uses:
```C++
printf(NEOFETCH_COLOR "%-*s", logo_width, (i < logo_height) ? neofetch_logo[i] : "");
```

Like this:
```C++
printf(NEOFETCH_COLOR "%-*s", logo_width, (i < logo_height) ? neofetch_logo[i] : "" NEOFETCH_INFO_COLOR);
```
to print the ascii art. (I made some custom ascii art, that's why I want to modify the colors.) And it turned out like this:
[image]

If it helps, I am using Momentum 8.0 as the base for my customized build. None of the modifications are in cli_ansi.h, so I am not sure that is doing anything, it may be how I applied the colors.

GameLord2011, 2024-12-02:
Note: The following code snippet
```C++
printf(NEOFETCH_COLOR "%-*s", logo_width, (i < logo_height) ? neofetch_logo[i] : "" ANSI_RESET NEOFETCH_INFO_COLOR);
```
produces the same result.

GameLord2011, 2024-12-02:
@Willy-JL @HaxSam?

Willy-JL, 2024-12-02:
Don't ping people, someone will reply when they have time and interest to do so. Patience is a virtue. Also you should learn what netiquette is.

GameLord2011, 2024-12-02:
Sry.

closed 2025-03-17 by 956MB