Next-Flip/Momentum-Firmware #317: 12 Hour Clock Formatting Issues

GonzalesGabe, 2024-12-08:
### Describe the bug.

When the 12 hour clock is enabled, the lock screen clock shows a leading zero for single-digit hours, and "12:XXam" is incorrectly shown as "00:XXam" 

### Reproduction

Make sure 12h clock is enabled
check lockscreen

### Anything else?

the leading zero issue is also present in the status bar clock, like the official firmware. If this is actually a system limitation, I understand it might not be fixable within Momentum Firmware

Willy-JL, 2024-12-08:
Personally I find 00:xx am to be more obvious when wanting to indicate that it is just past midnight. 12:xx am is ambiguous whether it means midnight or midday imo
Is there some standard saying it should be 12:xx am?

GonzalesGabe, 2024-12-08:
> Personally I find 00:xx am to be more obvious when wanting to indicate that it is just past midnight. 12:xx am is ambiguous whether it means midnight or midday imo Is there some standard saying it should be 12:xx am?

First of all that's a very fair point, I get what you mean, I do see

closed 2025-05-08 by WillyJL