Next-Flip/Momentum-Firmware #319: I found out how to change the cli prompt.

GameLord2011, 2024-12-11:
1. Change the "CLI_PROMPT," and the "CLI_PROMPT_LENGTH," variables accordingly.
2. Build the firmware.
3. Flash the firmware.
4. Clone "flipperdivices/qFlipper," or "Next-Flip/qFlipper," and download all standalone build dependencies.
5. Edit line 20 of "backend/flipperzero/rpc/stoprpcoperation.cpp," and line 19 of "backend/flipperzero/rpc/skipmotdoperation.cpp" to match the new prompt.
6. Build qFlipper.
7. Done!

GameLord2011, 2024-12-11:
Note, you will have to repeat steps 1-3 for Flipper firmware updates, and steps 4-6 for qFlipper updates, and for all computers you use qFlipper on.

Willy-JL, 2024-12-11:
Good job, but I don't understand why this is here. And it will break compatibility with other companion apps.

closed 2024-12-11 by Willy-JL