Next-Flip/Momentum-Firmware #320: Is it possible to re-add dummy mode?

GameLord2011, 2024-12-11:
My brothers love to steal my flipper zero to use all my amiibo's that I have on my flipper. The problem is that I want to make it to where they can only do that, because:

-They are clumsy.
-They will break some dumb setting, format my sd card and/or brick my flipper.
-And 173 other reasons I won't get into.

I also want to pin-lock it on dummy mode. Is there any way (stable) to re-implement dummy mode and add pin-lock functionality to it?
Since there is no more space on the top menu, I would be fine if you replaced Dark Mode or Lefty Mode or made it into a standalone plugin.

Willy-JL, 2024-12-11:
Adding dummy mode would require a significant amount of effort, it's not something you just "add back". Lots of other things now rely on it not existing. And we have no interest in adding it back. The issue you described can be fully solved with just setting a pin lock, in settings > desktop > pin setup, then locking with PIN code in the control center (press up from desktop)

closed 2024-12-11 by Willy-JL

GameLord2011, 2024-12-13:
Which things?