Next-Flip/Momentum-Firmware #321: u8x8 fonts in asset packs

GameLord2011, 2024-12-13:
### Description of the feature you're suggesting.

I noticed that the u8x8 part of the u8g2 library in the firmware dependencies. I am experimenting with fonts, and a while ago I accidentally put a u8x8 font in my asset pack, and it caused a build error. Is it possible to add u8x8 fonts to asset packs, if so, how?

956MB, 2024-12-25:
What file types are your font files you have in the asset pack, and what are the steps leading to the build error? Are you not loading the asset pack onto the SD card?

Willy-JL, 2024-12-25:
Asset packs use u8g2 font format, it's a C file that is converted to a binary blob that u8g2 can understand. u8x8 is another format and not one u8g2 can use from what I know (from what I understand u8g2 and u8x8 are 2 parts of the same library that work differently), so even if the asset packer could convert such fonts it wouldn't work afaik.

GameLord2011, 2024-12-26:
@956MB, I am using a custom asset pack that I put in the asset packs folder, I added a folder called "Fonts," and in it I added a file called "primary.c," in it I put in a font section from the U8G2 library. Once I accidentally put a u8x8 font section in there, and built it using the Build update bundle [release]
And during the build it said:
```
Traceback (most recent call last):
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\assets.py", line 395, in <module>
    Main()()
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\flipper\app.py", line 41, in __call__
    return_code = self.call()
                  ^^^^^^^^^^^
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\flipper\app.py", line 57, in call
    return self.args.func()
           ^^^^^^^^^^^^^^^^
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\assets.py", line 384, in packs
    asset_packer.pack(
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\asset_packer.py", line 200, in pack
    pack_font(font, packed / "Fonts" / font.name)
  File "C:\Users\*****\OneDrive\Documents\GitHub\Momentum-Firmware\scripts\asset_packer.py", line 112, in pack_font
    src.read_bytes().split(b' U8G2_FONT_SECTION("')[1].split(b'") =')[1].strip()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
scons: *** [build\f7-firmware-C\assets\compiled\asset_packs\Momentum\Anims\Kuronons_CFW_Momentum3_128x64\frame_2.bm] Error 1

********** FBT ERRORS **********
build\f7-firmware-C\assets\compiled\asset_packs\Momentum\Anims\Kuronons_CFW_Momentum3_128x64\frame_2.bm: Error 1
```
(error under kuronons asset pack becuase build tasks are asyncronous)

GameLord2011, 2024-12-26:
I just did a new boult becuase i dont have terminal logs

GameLord2011, 2024-12-26:
Please do not ask for the og logs becuase I uninstalled Ubuntu wsl and (tried) to replace it w/ kali linux, and my pc bluescreened while installing kali, had to reinstall ubuntu

Willy-JL, 2024-12-26:
@GameLord2011 as i said:
> Asset packs and flipper itself use u8g2 font format, it's a C file that the asset packer converts to a binary blob that u8g2 can understand. u8x8 is another format and not one u8g2 can use from what I know (from what I understand u8g2 and u8x8 are 2 parts of the same library that work differently), so even if the asset packer could convert such fonts it wouldn't work afaik.

closed 2025-01-23 by 956MB