Next-Flip/Momentum-Firmware #326: How to avoid losing the key when flashing the machine?

tianyi725, 2024-12-20:
### Description of the feature you're suggesting.

I have a question: How do you flash the core2?
Is it only core2 that can lose the key in the genuine version when flashing?
The normal "install from file" loading of tgz file flashing has no impact on the key.  

I want to know the brushing method of core2 to avoid the loss of key.
thankyou

(I didn't know where to ask, so I came here, you know)

Willy-JL, 2024-12-20:
You don't need to flash core2. Normal firmware update will flash core2 if the firmware for it changed or if it is corrupted

tianyi725, 2024-12-31:
> You don't need to flash core2. Normal firmware update will flash core2 if the firmware for it changed or if it is corrupted

I don't want to flash core2.
I'm just afraid that when I flash a third-party firmware (such as Momentum or others), I might accidentally flash core2, causing the loss of the genuine key.

Willy-JL, 2024-12-31:
No that's not the case don't worry. And actually installing any firmware, including official firmware, will flash core2 if the radio firmware is outdated. Flashing core2 just updates the Bluetooth firmware and other stm32 core logic, it doesn't touch what you're worried about. That's handled by the Secure Enclave, and that's never touched.

closed 2024-12-31 by Willy-JL