Next-Flip/Momentum-Firmware #328: Unknow error when installing the momentum firmware

Trojankill, 2024-12-22:
### Describe the bug.

When installing the Momentum firmware, it returns an error at 11% load. ( firmware version 1.1.2 program version 1.3.3 ) Firmware version momentum 008

### Reproduction

1 install from file 
2 select the file flipper-z-f7-update-mntm-008.tgz
3 install

### Logs

```Text
[qFlipper-20241222-122058.txt](https://github.com/user-attachments/files/18221366/qFlipper-20241222-122058.txt)
```

### Anything else?

I bought it recently and apart from formatting the SD card and updating the firmware, I did not do it.

Trojankill, 2024-12-22:
[image]
I read on the Internet that it has something to do with the driver DFU in FS mode

Willy-JL, 2024-12-22:
Sounds like you have both qflipper and the web installer open. Only one can work at a time. Close the browser to be sure when you use qflipper

Trojankill, 2024-12-22:
> Похоже, у вас открывается и qflipper, и веб-установщик. Вместе может работать только один из них. Закройте браузер, чтобы обеспечить надежность, когда используется qflipper

In the photo, I only used qflipper (the problem is that all the ways to install the momentum firmware on my computer don't work, and it takes a very long time to download on my phone).

Trojankill, 2024-12-22:
I solved the problem (the problem was in the sd card) I hope it helps someone.
The topic is closed

closed 2024-12-22 by Trojankill