Next-Flip/Momentum-Firmware #329: NRF24 Sniffer problem

pizzahaat, 2024-12-24:
### Describe the bug.

On the nrf24 sniffer application? whenever you press the middle button, to start, it buzzes and turns the light red instead of starting the sniffing process.

### Reproduction

1. connect nrf24 module 2. open sniffer application 3. press ok or middle button

### Logs

```Text
[7:50:37 PM][info]: app version: 1.8.1(586)
[7:50:37 PM][info]: log level: debug
[7:50:38 PM][info]: connecting
[7:50:38 PM][info]: session started
[7:50:41 PM][info]: connected
[7:50:41 PM][info]: synchronize
[7:50:41 PM][debug]: > request(storage(stat(/ext/Manifest)))
[7:50:41 PM][debug]: < response(storage(stat(162343)))
[7:50:41 PM][debug]: > request(system(property(devinfo.hardware.target)))
[7:50:42 PM][debug]: < response(system(property(hardware.target: 7)))
[7:50:42 PM][debug]: < response(ok)
[7:50:42 PM][debug]: > request(storage(read(/int/.region_data)))
[7:50:42 PM][debug]: < response(storage(file(58 bytes)))
[7:50:42 PM][debug]: > request(storage(info(/int)))
[7:50:42 PM][debug]: < response(storage(info(326 MiB / 119 GiB)))
[7:50:42 PM][debug]: > request(system(setDate(2024-12-24 00:50:41 +0000)))
[7:50:42 PM][debug]: < response(ok)
[7:50:42 PM][info]: syncing date: 1037 ms
[7:50:42 PM][debug]: > request(system(property(devinfo.firmware.api)))
[7:50:42 PM][debug]: < response(system(property(firmware.api.major: 78)))
[7:50:42 PM][debug]: < response(system(property(firmware.api.minor: 1)))
[7:50:42 PM][debug]: < response(ok)
[7:50:42 PM][debug]: > request(storage(info(/ext)))
[7:50:42 PM][debug]: < response(storage(info(326 MiB / 119 GiB)))
[7:50:42 PM][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[7:50:42 PM][debug]: < error(StorageError: File/Directory does not exist)
[7:50:42 PM][debug]: > request(storage(list(/ext/apps_manifests, true, 0))
[7:50:43 PM][debug]: < response(storage(list([nfc_maker.fim:797:8c4ee310b843d40279025e45741bb604), key_copier.fim:397:0fd42b100d2102bec62f720cb73ab4ed), nfc_rfid_detector.fim:399:74369c43f53392ee304b7bf06da68eee), tuning_fork.fim:375:2e0f1cf7ed9e87ffdb2b2d7551c3ce4a), snake20.fim:397:d5245b17893aed61d820777f35470110), findmy.fim:925:75cc46744b32803cc7104ecdb85f76aa)])))
[7:50:43 PM][debug]: > request(system(property(devinfo.firmware.origin.fork)))
[7:50:43 PM][debug]: < response(system(property(firmware.origin.fork: Momentum)))
[7:50:43 PM][debug]: < response(ok)
[7:50:43 PM][debug]: > request(storage(list(/ext, false, 0))
[7:50:43 PM][debug]: < response(storage(list([favorites.txt:0:), update, lfrfid, .int, Manifest:162343:), apps, apps_data, asset_packs])))
[7:50:43 PM][debug]: < response(storage(list([badusb, dolphin, ibutton_fuzzer, infrared, lfrfid_fuzzer, mifare_fuzzer, nfc, subghz])))
[7:50:43 PM][debug]: < response(storage(list([u2f, wav_player, ibutton, .tmp, apps_manifests, apps_assets, stress_test:1024:), subghz_remote])))
[7:50:43 PM][debug]: > request(storage(list(/ext/apps/NFC, false, 0))
[7:50:43 PM][debug]: < response(storage(list([cyborg_detector.fap:3688:), mfc_editor.fap:23928:), mfkey.fap:28748:), mifare_fuzzer.fap:17008:), nfc.fap:191536:), nfc_magic.fap:92404:), nfc_maker.fap:35312:), nfc_playlist.fap:33312:)])))
[7:50:43 PM][debug]: < response(storage(list([picopass.fap:117796:), seader.fap:168648:)])))
[7:50:43 PM][debug]: > request(system(property(devinfo.firmware.origin.git)))
[7:50:43 PM][debug]: < response(system(property(firmware.origin.git: https://github.com/Next-Flip/Momentum-Firmware)))
[7:50:43 PM][debug]: < response(ok)
[7:50:43 PM][debug]: > request(storage(list(/ext/subghz, true, 10485760))
[7:50:43 PM][debug]: < response(storage(list([Tesla, assets, playlist, remote, Garage-door-house1.sub:194:a86a277e7b54f80b621453d458d2c5ab), Fan_light.sub:187:cdd6148f5c4a28e7a6d4546cfbed3819), Fan_max.sub:185:692a2a57310289dfd408b0c92e14da1d), Lowes])))
[7:50:43 PM][debug]: < response(storage(list([CVS, Walgreens, Gas signs, Princeton_2024-12-14_22,21,33.sub:213:726011cdd8426900a8f6ed6640391dac)])))
[7:50:43 PM][debug]: > request(storage(list(/ext/apps/Tools, false, 0))
[7:50:43 PM][debug]: < response(storage(list([bad_kb.fap:53148:), barcode_app.fap:36424:), caesar_cipher.fap:3776:), calculator.fap:50464:), cli_gui.fap:8084:), cntdown_tim.fap:6848:), counter.fap:3792:), dtmf_dolphin.fap:29980:)])))
[7:50:43 PM][debug]: < response(storage(list([flipbip.fap:132060:), flipp_pomodoro.fap:21884:), hex_editor.fap:6020:), hex_viewer.fap:20272:), key_copier.fap:24532:), multi_converter.fap:14200:), nfc_rfid_detector.fap:11964:), nightstand.fap:7388:)])))
[7:50:43 PM][debug]: < response(storage(list([passgen.fap:6796:), qrcode.fap:15512:), quac.fap:57032:), text_viewer.fap:6036:), tone_gen.fap:24404:), totp.fap:104752:), upython.fap:267136:), xbox_controller.fap:7180:)])))
[7:50:43 PM][debug]: < response(storage(list([meal_pager.fap:48848:), subghz_remote.fap:47272:), math.js:3071:)])))
[7:50:43 PM][debug]: > request(storage(list(/ext/lfrfid, true, 10485760))
[7:50:43 PM][debug]: < response(storage(list([])))
[7:50:43 PM][debug]: > request(storage(list(/ext/apps/Media, false, 0))
[7:50:44 PM][debug]: < response(storage(list([bpm_tapper.fap:6576:), etch.fap:5280:), flizzer_tracker.fap:66908:), metronome.fap:11548:), morse_code.fap:8440:), music_player.fap:12724:), paint.fap:3020:), text2sam.fap:29100:)])))
[7:50:44 PM][debug]: < response(storage(list([tuning_fork.fap:14068:), wav_player.fap:14452:)])))
[7:50:44 PM][debug]: > request(storage(list(/ext/nfc, true, 10485760))
[7:50:44 PM][debug]: < response(storage(list([RickRoll.nfc:5323:5bd26508fc7103d22cef4f41d18a73d7), assets, Jack Land snowshoe.nfc:1517:52204d491c64074f91cf73f91d955f78)])))
[7:50:44 PM][debug]: > request(storage(list(/ext/apps/Games, false, 0))
[7:50:44 PM][debug]: < response(storage(list([4inrow.fap:5780:), air_arkanoid.fap:42400:), air_labyrinth.fap:29700:), arkanoid.fap:5936:), asteroids.fap:33744:), blackjack.fap:25168:), bomberduck.fap:12912:), chess.fap:62164:)])))
[7:50:44 PM][debug]: < response(storage(list([color_guess.fap:29232:), doom.fap:48980:), flappy_bird.fap:9764:), flipper_pong.fap:4488:), game15.fap:9752:), game_2048.fap:11632:), gameoflife.fap:3992:), heap_defence.fap:18228:)])))
[... 176 lines cut ...]
[7:52:00 PM][debug]: < response(storage(file(512 bytes)))
[7:52:00 PM][debug]: < response(storage(file(512 bytes)))
[7:52:00 PM][debug]: < response(storage(file(512 bytes)))
[7:52:00 PM][debug]: < response(storage(file(512 bytes)))
[7:52:00 PM][debug]: < response(storage(file(512 bytes)))
[7:52:00 PM][debug]: < response(storage(file(500 bytes)))
[7:57:04 PM][debug]: > request(storage(list(/ext/apps/GPIO/NRF24, false, 0))
[7:57:04 PM][debug]: < response(storage(list([nrf24batch.fap:42980:), nrf24channelscanner.fap:8660:), nrf24mousejacker.fap:22340:), nrf24scan.fap:32796:), nrf24sniff.fap:16884:)])))
[7:57:04 PM][debug]: > request(storage(list(/ext/apps/GPIO, false, 0))
[7:57:04 PM][debug]: < response(storage(list([Debug, ESP, FlipperHTTP, MALVEKE, MAYHEM, NRF24, Sensors, VGM])))
[7:57:04 PM][debug]: < response(storage(list([air_mouse.fap:106096:), coleco.fap:13532:), flashlight.fap:3676:), flipperscope.fap:22880:), gpio.fap:37020:), gpio_badge.fap:4676:), gpio_controller.fap:13840:), gpio_reader_a.fap:4724:)])))
[7:57:04 PM][debug]: < response(storage(list([gpio_reader_b.fap:26956:), gpio_sentry_safe.fap:4720:), gps_nmea.fap:14384:), magspoof.fap:48104:), pokemon.fap:182256:), signal_generator.fap:11416:), timelapse.fap:10068:), uart_terminal.fap:24384:)])))
[7:57:04 PM][debug]: < response(storage(list([wii_ec_anal.fap:34188:), wire_tester.fap:3944:), FZ_Marauder_v2.8.zip:0:)])))
[7:57:05 PM][debug]: > request(storage(list(/ext/apps, false, 0))
[7:57:05 PM][debug]: < response(storage(list([Bluetooth, GPIO, Games, Infrared, Media, NFC, RFID, Scripts])))
[7:57:05 PM][debug]: < response(storage(list([Sub-GHz, Tools, USB, assets, iButton])))
[7:57:06 PM][debug]: > request(storage(list(/ext, false, 0))
[7:57:06 PM][debug]: < response(storage(list([favorites.txt:0:), update, lfrfid, .int, Manifest:162343:), apps, apps_data, asset_packs])))
[7:57:06 PM][debug]: < response(storage(list([badusb, dolphin, ibutton_fuzzer, infrared, lfrfid_fuzzer, mifare_fuzzer, nfc, subghz])))
[7:57:06 PM][debug]: < response(storage(list([u2f, wav_player, ibutton, .tmp, apps_manifests, apps_assets, stress_test:1024:), subghz_remote])))
[7:57:44 PM][debug]: > request(storage(list(/ext, false, 0))
[7:57:44 PM][debug]: < response(storage(list([favorites.txt:0:), update, lfrfid, .int, Manifest:162343:), apps, apps_data, asset_packs])))
[7:57:44 PM][debug]: < response(storage(list([badusb, dolphin, ibutton_fuzzer, infrared, lfrfid_fuzzer, mifare_fuzzer, nfc, subghz])))
[7:57:44 PM][debug]: < response(storage(list([u2f, wav_player, ibutton, .tmp, apps_manifests, apps_assets, stress_test:1024:), subghz_remote])))
```

pizzahaat, 2024-12-24:
also on different apps if you go onto the app with the thing unplugged, then it will say no NRF 24 attached but then on other apps, it says NRF 24L01 +R/ERROR!

closed 2024-12-24 by pizzahaat

Willy-JL, 2024-12-24:
the logs you provided are of qflipper, not flipper. anyway, you probably changed momentum > protocols > gpio pins > nrf24, without knowing what it does.