Next-Flip/Momentum-Firmware #330: Lockscreen: Allow RPC while locked separately for Bluetooth (BLE) and USB

Msprg, 2024-12-24:
### Describe the enhancement you're suggesting.

It seems that the option “Allow RPC while locked” in the momentum app affects both USB and BLE connections. However, in other custom firmwares, it seems that locked flipper has USB connectivity disabled, but it's still possible to connect and control flipper by the android / iOS companion apps. This is useful as a convenience, as well as to find locked Flipper by playing alert sound, for example. Since in my opinion the BLE connection is a little more secure / trustworthy due to BLE pairing as opposed to USB, it would be nice if we could allow RPC while locked for BLE and USB separately.
(Or this might also just be a user error or a bug, since I haven't found any other issues regarding this IMO basic functionality)

### Anything else?

Thank you for your hard work on this wonderful CFW, and happy holidays~!

956MB, 2025-01-13:
https://github.com/Next-Flip/Momentum-Firmware/compare/37d8bf76...dbbecd9e1fbf4449af56f1eec0c69979e7b8538f

closed 2025-01-13 by 956MB

Msprg, 2025-01-13:
Thank you so much~! ❤️