Next-Flip/Momentum-Firmware #332: why does it make my micro sd's not work

dovydasa2, 2024-12-25:
### Describe the bug.

so when i use my flipper zero and try to turn it off  by holding left arrow and back button it just makes my Sd card's not work and two of them are just not working right now because of this.

### Reproduction

1.hold left arrow and back button 
2. it turns off for a few seconds 
3. your Sd card magickly doesn't work

dovydasa2, 2024-12-25:
pls help

956MB, 2024-12-25:
> pls help

Couple of questions:

1. SD brand/size/format?
2. You had momentum flashed and working fine until rebooting with *left*+*back*?
3. Anything visible on screen during/after the reboot attempt?
4. When the flipper is off and you try turning it on again with *back* for 3 seconds, nothing happens at all?
5. Does the SD card show up and/or look fine plugged into your computer? Is it corrupted?

dovydasa2, 2024-12-25:
1.i don't know
2.it was working fine
3.nothing during reboot but there was the screen that said mounting Sd card failed immediately 
4. well the thing is that it just didn't turn off it rebooted immediately
5. it doesn't let my look and I'm going to add a screen shot just so you can see 
[image]
btw i literally tried to format it every ware from my Nintendo to the pc and it always says that its unable

Anonymous3-a, 2024-12-25:
Maybe a video of the MicroSD card being put in and left+back would help?

dovydasa2, 2024-12-25:
Here it is https://we.tl/t-mV57C97qn9

Anonymous3-a, 2024-12-25:
I meant what it looks like when the MicroSD card stops working. It also seems like you aren't verifying the MicroSD cards beforehand on a computer. In addition, it looks like a non-name brand MicroSD card. Those do not always support the low-power interface the Flipper Zero uses, so maybe there's something to do with that? The problem could also be something in the slot, like a tiny bit of metal that got in and is shorting some contacts.

Try:

 - Making sure the MicroSD chip worked before use in the flipper
 - Blowing into the MicroSD slot (compressed air optional)
 - Using a name-brand MicroSD chip (SanDisk works good in my experience)
 - If you are able to read the chip in the flipper before it stops working, maybe try updating (upload the update file to the card, and update from the file browser)

dovydasa2, 2024-12-25:
Ama buy a better sd tomorrow

Willy-JL, 2024-12-25:
from that video, one of the sd cards is corrupted, and the other one doesnt have firmware files on it. choose one of the 2 sd cards, plug it into flipper, go to settings > storage > format sd card, format it there, then reinstall the firmware either webupdater, or qflipper, or mobile app.

dovydasa2, 2024-12-25:
Tried formating both didn't work

Willy-JL, 2024-12-25:
did you reinstall the firmware as i told you to do?

dovydasa2, 2024-12-25:
Hpw to do that and how do you know that one is corrupted and the other one dosent have frimware

Willy-JL, 2024-12-25:
the flipper literally tells you. but you refuse to read.
it tells you that the app cannot run, and you need to install firmware.
the other one tells you that it is a bad sd card, so either badly formatted or corrupted.
in any case, both of those are fixed by format + reinstall firmware.
reinstall firmware just means install firmware again. currently, its liek you dont have part of the firmware. just install using any of the supported installation methods. qflipper, webupdater, mobile app... like i already told you.

dovydasa2, 2024-12-26:
i got a new Sd ama reply in a minute if the glitch still works

dovydasa2, 2024-12-26:
everythings works just fine thanks for the help