Next-Flip/Momentum-Firmware #338: Battery percentage on MNTM menu style is off by one percent!

GameLord2011, 2024-12-31:
### Describe the bug.

The battery percent on the mntm menu style is one less then the percent shown on the desktop

### Reproduction

1. Turn on flipper zero.
2. Switch menu style to mntm.

### Target

Firmware, Hardware

### Anything else?

my suggestion for a fix is to change line 409 of `menu.c` to:
```C
snprintf(bat_display, sizeof(bat_display), "%d%%", battery_percent + 1);
```

956MB, 2024-12-31:
Probably better to use the function that's already there for that than to add 1, I think. :)

956MB, 2025-01-13:
https://github.com/Next-Flip/Momentum-Firmware/compare/37d8bf76...dbbecd9e1fbf4449af56f1eec0c69979e7b8538f

closed 2025-01-13 by 956MB