Next-Flip/Momentum-Firmware #340: Charge cap not working.

GameLord2011, 2024-12-31:
### Describe the bug.

I have my flipper's charge cap set to 95 percent, and while updating it charged to 98 percent!

### Reproduction

1. Go to momentum settings
2. set the charge cap to 95%
3. get your flipper to 94 percent
4. update the firmware

### Target

Firmware, Hardware

956MB, 2024-12-31:
I've observed the same having flashed firmware again and again, my 80% cap is always exceeded. Perhaps due to the device getting power in the interval of updating and in that time it's not checking the cap? Not sure.

Willy-JL, 2024-12-31:
It's not exactly clear how it works (not that I've tried with all my might to find out), but from what I saw even when charge is suppressed (method that charge cap uses, aswell as some apps like subghz as they need a stable power line without fluctuations that charging causes) it's not discharging like you'd expect... unplugged it shows charge of -6ma or whatever, charging shows +whatever ma, and plugged with charge suppressed it shows charge 0ma... almost like it is charging but only enough to offset the amount it's drawing. That might explain why, it's "suppressed" but a very time amount is still being accumulated over time? If so, no clue if it would be a software or hardware thing. And then again, not even sure if this is what's actually happening, maybe that 0ma is just a problem in display but it's actually discharging, I don't know.
Or it could be charging during reboots yeah.

GameLord2011, 2025-01-01:
It's not just charging just during reboots; it's also charging when I get files through qFlipper and download or update apps

956MB, 2025-01-01:
> It's not just charging just during reboots; it's also charging when I get files through qFlipper and download or update apps

Well yes, but we count those as 'normal operation' and the cap should be working correctly for them, right?