Next-Flip/Momentum-Firmware #352: uPython crashes - Error: Out of Memory

AztecCodes, 2025-01-14:
### Describe the bug.

When I try to start **uPython** it will crash after a short loading period.

**The error message:**
> Error: Out of Memory
> Not enough RAM to run the app. Please reboot the device.

I never managed to start **uPython**.

**Device Information**
- Firmware: MNTM-DEV E185F322
- Model: FZ.1
- SD-Card Format: ExFAT

### Reproduction

1. Turn on your Flipper Zero
2.  Navigate to Apps and open
3. Open the "Tools" folder
4. Open the uPython application
5. Now the device should show the Error: Out of Memory exception

Willy-JL, 2025-01-14:
here: https://downloadmoreram.com/

Willy-JL, 2025-01-14:
flipper has 256kb of ram and its never enough. we have extra features and it uses a tiny bit more that tips it over the point where its more likely to not work than to work. still, same issue on all firmwares with maybe slightly different chances of success

AztecCodes, 2025-01-14:
@Willy-JL Why have uPython installed with you can't start it?

Willy-JL, 2025-01-14:
you can start it, but its unlikely.

AztecCodes, 2025-01-14:
I see, sadly it never worked for me yet. Also using the Momentum asset pack to save RAM and tried to start uPython after reboot.

Willy-JL, 2025-01-14:
the momentum asset pack is among the heaviest in terms of ram usage. thats probably your issue. you probably go from 10% chance of opening to 0.1% chance of opening by using a large asset pack like that

AztecCodes, 2025-01-14:
@Willy-JL I thought the momentum one is the smallest asset pack. It literally only has one short animation.
Most other asset packs I see have mutliple animations and also stuff like custom icons.

Willy-JL, 2025-01-14:
animations are unloaded from ram before opening apps, icons and fonts are not. momentum pack has the most icons of any pack. the upython app on momentum also fully unloads asset packs icons and fonts before opening, but the impact of things being loaded before is that memory can become fragmented so in some cases thats not enough anyway.

Willy-JL, 2025-01-14:
default one, or the blank screen from the website if you want to go overkill

AztecCodes, 2025-01-14:
@Willy-JL  Thank you! I will close the ticket then I guess.

closed 2025-01-14 by AztecCodes