Next-Flip/Momentum-Firmware #371: Frequent crashes when GPS attached to serial

zenmetsu, 2025-02-06:
### Describe the bug.

With an L76K GPS module attached to serial, momentum firmware seems to crash while on the home screen (whilst playing animations), and also it appears to crash frequently when in the sub-ghz frequency analyzer.  What is strange is that I can run the GPS app for hours on end without issue, but upon exiting it will often hang the flipper while in the menus.

I have ruled out a power issue by attaching just ground and 3v3 between the GPS module and the Flipper.  The issue is only apparent when the serial line is attached to pin 14 of the Flipper.  I have scoped the GPS output to verify that it is not sending out-of-spec logic levels (it is 0V to 3.05V)  I suspect that the data coming in via serial is somehow causing resource issues on the Flipper when running the Momentum firmware.

### Reproduction

1. Attach a 3v3 serial device's data output to pin 14.
2. Power up the Flipper.
3. Enter the Frequency Analyzer within the sub-ghz menu and allow it to run for a while.
4. Flipper will usually hang within a few seconds.

I have noticed that by going into the Radio Settings option within Sub-GHz and setting GPS from Off to the baud rate of the GPS, the issue is reduced in frequency, however it still occurs.

### Logs

```Text

```

Willy-JL, 2025-02-06:
you have "settings > expansion modules > listen uart" enabled, and set to the channel that the gps module uses. unfortunately this has been a known bug with very little information for a very long time.

expansion modules are a way for gpio modules to interact with flipper rpc without needing a custom app, currently the only device that uses it is the video game module but nothing stops others from developing devices using said protocol. it works by reading all data received on the selected uart channel and trying to parse it as expansion protocol data, resetting if it is not. somewhere in that black box theres a bug that can cause it to freeze. ive tried multiple times to debug it but its seemingly random and without notice. its also not a momentum issue, it will happen on all firmwares when expansion service is enabled.

you dont get it inside the gps app because there, the app is reading data from the gps, so the expansion service is disabled. you get it in subghz frequency analyzer because expansion service works in background, though subghz does have support for gps modules and it would therefore not happen in subghz if you enabled "subghz > radio settings > gps baudrate". or you can just:
- unplug the thing
- disable expansion listen uart

zenmetsu, 2025-02-06:
I suppose i am just an idiot.  thanks for the prompt reply.  I'll go ahead and close this since it has addressed my issue.

closed 2025-02-06 by zenmetsu

Willy-JL, 2025-02-06:
not an idiot, it is a bug afterall. and in hindsight, even though tehrabbitt (from rabbitlabs) and i found this bug long ago (almost a year) and maybe 2-3 people total ever noticed it so far, we never reported it properly to upstream, so you didnt have much of a way to know this.

though i see it has been reported now and assigned, might be fixed eventually
https://github.com/flipperdevices/flipperzero-firmware/issues/4081

zenmetsu, 2025-02-06:
I can look into the bug since I can semi-reliably trigger it.