Next-Flip/Momentum-Firmware #373: Question about FlipperNested / Mifare Nested. Removed? Why?

ispjstfu, 2025-02-09:
Was talking to a friend about cracking nonces from a card/key fob and not just a reader using Mifare Nested/FlipperNested (https://github.com/AloneLiberty/FlipperNested) right after he updated to MNTM-009 while I was still at MNTM-007 and he said he couldn't find that app under NFC > Mifare Nested where I had it.

After backing up my SD card and all apps I too updated and indeed it is gone. I guess because it was considered broken or superfluous? Couldn't find anything about it in the release notes.

Anyway, I found that app useful recently so I am curious as to why it was removed, and if there are any alternatives to reading nonces from a card/key fob and not just a reader? Using Extract MF Keys does not seem to do anything when used that way, only against a reader.

If not, is there any harm in adding it back? I saved the .fap file and it still seems to work for my friend who I sent it to before updating myself.

Thanks!

Willy-JL, 2025-02-09:
I spend time writing changelogs exactly for this ;)

FlipperNested (the flipper app) never cracked any nonces, it only collected them. That whole functionality is now fully built into the main NFC app, during read. And you can now also crack nested nonces on flipper, using MFKey.

There's also a lot more magic, noproto spend a whole year collecting and doing research about these attacks, aswell as backdoor keys, PRNG prediction and dictionary attack optimizations (it's to the point where he was writing code at millisecond precision to synchronize with the tag), the whole process is about 1000x faster and almost fully automated.

Just use NFC > Read, and MFKey, that will cover 90% of tags with just 2 steps, most even just with NFC app that would've previously required cracking nonces. The only case where you'll need a PC is for hard nested nonces, in which case the NFC app will show "(Hard)" text at the top while reading as it collects these nonces, and you'll need a PC to crack these. Noproto also very kindly put up a website to crack hard nested nonces on his powerful hardware while waiting for the flipper mobile app to get support for cracking hard nested nonces: https://flipperzero.club/hardnested/

The current intended process is this: https://flipper.wiki/mifareclassic/

closed 2025-02-09 by Willy-JL

ispjstfu, 2025-02-09:
Thank you so much for the detailed and kind answer! I am quite new to this but I love learning about it all. Will definitely try it this way instead. Keep up the amazing work! 💯

tommiv, 2026-01-17:
@WillyJL  alright, thanks, will try.