Next-Flip/Momentum-Firmware #377: Sub-GHz - Frequency Analyzer not using external CC1101 module for scanning

chengetk, 2025-02-28:
### Describe the bug.

1. Frequency Analyzer not using external CC1101 module for scanning when select 'Sub-GHz' -> 'Frequency Analyzer' from the main menu.
2. Frequency Analyzer only uses the external CC1101 module when select from 'Apps' -> 'Sub-GHz' -> 'Frequency Analyzer'

### Reproduction

1. Connect CC1101 module to Flipper Zero
2. Select 'Sub-GHz' from menu
3. Select 'Frequency Analyzer'
4. Flipper Zero uses internal antenna module to scan even there is an external CC1101 module

### Logs

```Text

```

### Anything else?

Firmware MNTM-009 23-01-2025

Willy-JL, 2025-02-28:
this is intentional, external cc1101 do not have the required radio hardware to make frequency analyzer work correctly and accurately. and if they did, they would have the same range as the internal module. the results you get with external modules are inaccurate and incorrect.

closed 2025-02-28 by Willy-JL

Willy-JL, 2025-02-28:
it was explained well in the changelogs