Next-Flip/Momentum-Firmware #392: Help compile app PLEASE

pigmen11, 2025-04-04:
Hello! I've been trying my hardest to figure out how to create a simple scrolling app with Bluetooth but failing. Basically all I want to create is an app just like the mouse clicker app, instead I would just need it to scroll upwards continuously upon toggle, is this possible?

Would the following code below help at all? 

#include <furi.h>
#include <gui/gui.h>
#include <input/input.h>
#include <ble/ble.h>

#define TAG "ScrollToggleApp"

static bool is_scrolling = false;
static FuriThread* scroll_thread = NULL;

static void scroll_thread_callback(void* arg) {
    UNUSED(arg);
    while(is_scrolling) {
        ble_hid_mouse_scroll(1); // Scroll up by 1 unit
        furi_delay_ms(100);      // Scroll every 100ms
    }
}

static void input_callback(InputEvent* event, void* context) {
    UNUSED(context);
    if(event->type == InputTypePress) {
        if(event->key == InputKeyOk) {
            is_scrolling = !is_scrolling;
            if(is_scrolling && !scroll_thread) {
                scroll_thread = furi_thread_alloc_ex(TAG, 1024, scroll_thread_callback, NULL);
                furi_thread_start(scroll_thread);
            } else if(!is_scrolling && scroll_thread) {
                furi_thread_free(scroll_thread);
                scroll_thread = NULL;
            }
        } else if(event->key == InputKeyBack) {
            is_scrolling = false;
            if(scroll_thread) {
                furi_thread_free(scroll_thread);
                scroll_thread = NULL;
            }
            furi_record_close("gui");
            ble_hid_service_stop();
        }
    }
}

int32_t scroll_toggle_app(void* p) {
    UNUSED(p);
    FURI_LOG_I(TAG, "Starting Scroll Toggle App");

    ble_hid_service_start();

    Gui* gui = furi_record_open("gui");
    ViewPort* view_port = view_port_alloc();
    gui_add_view_port(gui, view_port, GuiLayerFullscreen);
    view_port_input_callback_set(view_port, input_callback, NULL);

    view_port_draw_callback_set(view_port, [](Canvas* canvas, void* ctx) {
        UNUSED(ctx);
        canvas_set_font(canvas, FontPrimary);
        canvas_draw_str(canvas, 10, 20, "Scroll Toggle");
        canvas_draw_str(canvas, 10, 40, "OK: Start/Stop");
        canvas_draw_str(canvas, 10, 50, "BACK: Exit");
    }, NULL);

    furi_thread_yield();
    return 0;
}

pigmen11, 2025-04-04:
I think I just need to compile into a FAP file... I hinestly don’t know what I'm doing

Willy-JL, 2025-04-04:
AI will not help you code. it will make your life miserable. learn to code, look at examples, but dont use AI.

closed 2025-04-04 by Willy-JL

pigmen11, 2025-04-04:
Haha thank you, I hateeee using Ai to code it's the worst😭😂 I have no idea where to even begin making the FAP file